realsenseai/librealsense #2038: Errors installing RealSense D435 on Ubuntu 16.04 LTS

alberto-tellaeche, 2018-07-11:
Dear all;

Yesterday I received my RealSense D435. I want to use it with Ubuntu 16.04.
After following direct instructions fron this repository, I obtain the following errors.

1.- Repository keys are not obtained:
```
alberto@alberto-OptiPlex-9010:~$ sudo apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE
Executing: /tmp/tmp.D9mZmCOxc4/gpg.1.sh --keyserver
keys.gnupg.net
--recv-key
C8B3A55A6F3EFCDE
gpg: requesting key 6F3EFCDE from hkp server keys.gnupg.net
gpg: keyserver timed out
gpg: keyserver receive failed: keyserver error

```
I have tried to install packages without verifying them, librealsense2-dkms and librealsense2-utils. It seems they hace installed everything ok.

2.- My second error is when trying to use the viewer:
```
alberto@alberto-OptiPlex-9010:~$ realsense-viewer 
 11/07 13:25:00,087 WARNING [139772276279936] (types.cpp:57) API version mismatch: librealsense.so was compiled with API version 2.11.1 but the application was compiled with 2.12.0! Make sure correct version of the library is installed (make install)
 11/07 13:25:00,102 WARNING [139772276279936] (rs.cpp:204) null pointer passed for argument "context"
RealSense error calling rs2_create_context(api_version:21200):
    API version mismatch: librealsense.so was compiled with API version 2.11.1 but the application was compiled with 2.12.0! Make sure correct version of the library is installed (make install)
alberto@alberto-OptiPlex-9010:~$ 
```

Can you provide any solution to this?
Best,
Alberto

| Required Info                    
|---------------------------------------------------------------------------- |
Camera Model                       D435| 
 Firmware Version                   Unable to open RealSense Viewer 
 Operating System & Version  Ubuntu 16.04 LTS
 Kernel Version (Linux Only)    4.13.0-45-generic                                      
 Platform                                  PC
 SDK Version                           development downloaded from repository
 Language                              C/C++            
 Segment			               Robotics

dorodnic, 2018-07-11:
Hi @alberto-tellaeche 
Sorry for the inconvenience you experience. 
Regarding the first point - it seems you are running behind firewall / proxy. Please check your network settings to ensure `keys.gnupg.net` is reachable. Also, please take a look at #1276 for potential solution.
Regarding the second point, as the error suggests, you seem to have librealsense 2.11.1 already installed, most likely as a result of `sudo make install`. If you have previously built the library from source, navigate to `build` folder and run `sudo make uninstall`. Alternatively, you can look under `/usr/local/bin`, `/usr/local/lib` and `/usr/local/include` and remove any remains of version 2.11.1.

alberto-tellaeche, 2018-07-12:
Hi Sergey and Realsense team,

These are my findings related to the problem attached below:

1.- To get the keys, I have followed instructions of #1276 (https://lodge.glasgownet.com/2012/07/11/apt-key-from-behind-a-firewall/)

apt-key from behind a firewall – The Lodge<https://lodge.glasgownet.com/2012/07/11/apt-key-from-behind-a-firewall/>
lodge.glasgownet.com
At work we’re pretty heavily firewalled. This means that outbound requests on funny ports are often firewalled off, and systems that rely on such ports will time out.

I still get errors:

alberto@alberto-OptiPlex-9010:~$ sudo apt-key adv --keyserver hkp://keys.gnupg.net:80 --recv-key C8B3A55A6F3EFCDE
[sudo] password for alberto:
Executing: /tmp/tmp.Q7ekmYL6E3/gpg.1.sh --keyserver
hkp://keys.gnupg.net:80
--recv-key
C8B3A55A6F3EFCDE
gpg: requesting key 6F3EFCDE from hkp server keys.gnupg.net
gpgkeys: key C8B3A55A6F3EFCDE can't be retrieved
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0
gpg: keyserver communications error: keyserver helper general error
gpg: keyserver communications error: unknown pubkey algorithm
gpg: keyserver receive failed: unknown pubkey algorithm
alberto@alberto-OptiPlex-9010:~$

I have retrieved many times keys for other software ( such as ROS, for robotics), and I haven' t had any problem so far.

2.- About library versions, I also use Nuitrack, and it includes a version of librealsense that is maybe the one causing problems:

alberto@alberto-OptiPlex-9010:~$ sudo updatedb
[sudo] password for alberto:
alberto@alberto-OptiPlex-9010:~$ locate realsense
/etc/apt/sources.list.d/realsense-public.list
/home/alberto/.cache/upstart/update-notifier-crash-_var_crash_librealsense2-dkms.0.crash.log.1.gz
/usr/local/lib/nuitrack/librealsense2.so.2
/var/lib/apt/lists/realsense-hw-public.s3.amazonaws.com_Debian_apt-repo_dists_xenial_Release
/var/lib/apt/lists/realsense-hw-public.s3.amazonaws.com_Debian_apt-repo_dists_xenial_main_binary-amd64_Packages
/var/lib/apt/lists/realsense-hw-public.s3.amazonaws.com_Debian_apt-repo_dists_xenial_main_binary-i386_Packages
alberto@alberto-OptiPlex-9010:~$

Is there a method for the two library versions to co-exist?

Thank you very much in advance,

Best,

Alberto

RealSense-Customer-Engineering, 2018-07-19:
[Realsense Customer Engineering Team Comment]
hi @alberto-tellaeche,

for your 1), i got same result with you for your command <code>sudo apt-key adv --keyserver hkp://keys.gnupg.net:80 --recv-key C8B3A55A6F3EFCDE</code> and ok to get the pacakge with <code> sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 (hkp://keys.gnupg.net:80) --recv-key C8B3A55A6F3EFCDE</code>

for 2), we are improving version by version, so would suggest you to use the latest version.

closed 2018-08-14 by RealSense-Customer-Engineering