realsenseai/librealsense #2042: on MacOS Corrupted Color Frame buffer when switching devices

obviousjim, 2018-07-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |   MacOS High Sierra 10.13.4  |  
| Platform                                 | PC |
| SDK Version                            |  2.13.0                          | 
| Language                            |  C                       | 
| Segment			|  VR/AR                          | 

### Issue Description

Sometimes when running and changing devices, the frames received from the RealSense API have a stride offset and or discoloration (green and magenta). We're seeing this in our own application, as well as RealSense Viewer. See screenshot below:

[image]

Any clues?

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
hi @obviousjim,

I never see this before, and you mentioned this happens sometimes, do you have some process to reproduce this?
and would like to know which version is your realsense viewer?
thanks.

napurvey, 2018-07-12:
Hi guys,

Picking this up from the initial report from @obviousjim.

The repro on this problem involves us running our own capture application with a Realsense D415 sensor (although could be happening on D435 too), waiting until the Realsense SDK locks up or crashes then seeing the result when starting the viewer app without unplugging and plugging the sensor back in. We also see the resulting problem (mismatched stride, discolorization of incoming frames) in our own capture app but confirmed it happens in the viewer app also, indicating the sensor must be in a bad state from the previous crash or lockup (from my POV).

We are seeing this most prevalently whilst testing on a Macbook Air with sdk v2.13 in our App and the viewer and latest matching sensor firmware v5.9.14.

This does not happen on every crash of the SDK, so is not reproducable every run but is a critical bug for our product.

The Realsense viewer version is v2.13.0 inline with the SDK version and freshly pulled and compiled from master.

Thanks!

Neil

napurvey, 2018-07-12:
Also another note; before upgrading the firmware and SDK it was also happening, we were previously on SDK v2.11.

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
hi @napurvey,

i see you have another ticket #2057, would like to know this is separate issue or the same issue?
thanks.

napurvey, 2018-07-23:
This is a diffrerent issue.

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
hi @napurvey ,
got you, but i still can't reproduce this issue so would like to confirm my reproduce steps if ok:
1. connect D435 and start streaming with Realsense viewer (IR+Depth+RGB)
2, turn on/off stereo module (IR+Depth) repeatedly, until realsense-viewer is crash.
3. don't unplug D435, restart realsense-viewer, repeat step.1 to see if issue reproduced.
4. if not reproduced, repeat step 2 to step 4.

i tested this like 10 runs and see no stride error, discolorization, but i do see frame not received issue.

napurvey, 2018-07-26:
Hi,

Unfortunately, this is not as frequent as the other instability issues we have seen although they are caused by getting the sensor into a bad state by reproducing the instabilities shown in issue #2057 then restarting the viewer app and the streams will display the stride / dis-colorization we see.

It may need rigorous testing to make this issue happen.

Thanks.

RealSense-Customer-Engineering, 2018-07-27:
[Realsense Customer Engineering Team Comment]
hi @napurvey,

got you, will keep an eye on this issue while I am working on #2057, and I think after stability issue fixed, this issue should be also fixed.

napurvey, 2018-09-25:
Thanks @dorodnic , we'll check it out.

RealSense-Customer-Engineering, 2018-10-16:
[Realsense Customer Engineering Team Comment]
Hi @napurvey,

Still any issue based on 2.16.1?