realsenseai/librealsense #2043: DS5 group_devices is empty, rame metadata isn't available! Warnings

fiorano10, 2018-07-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSense-Customer-Engineering, 2018-07-27:
[Realsense Customer Engineering Team Comment]
Hi @fiorano10,

Any update?
By the way, it looks your rs_camera.launch has some modification.

fiorano10, 2018-07-27:
@RealSense-Customer-Engineering 
Yes, a few changes were made to the launch file. I haven't had a chance to downgrade the kernel yet, will update you within a week.

Thanks!

RealSense-Customer-Engineering, 2018-08-06:
[Realsense Customer Engineering Team Comment]
Hi @fiorano10,

any update?

fiorano10, 2018-08-07:
@RealSense-Customer-Engineering , Updating the camera firmware, and the SDK to 05.09.14.00 seems to have fixed the issue.
Thanks!

fiorano10, 2018-08-08:
@RealSense-Customer-Engineering, the issue is back. Is there any other way, than downgrading the kernel?

RealSense-Customer-Engineering, 2018-08-13:
[Realsense Customer Engineering Team Comment]
Hi @fiorano10,

This issue is similar with #2029. Will further check this internally.

RealSense-Customer-Engineering, 2018-08-21:
[Realsense Customer Engineering Team Comment]
Hi @fiorano10,

A possible workaround is to have an simple app to reset the camera once you have the stream error.  Then, you can re-run the launch file.
The code like below. You can have a try. Also, look for if there is a better way to resolve this in the ROS wrapper.

rs2::pipeline pipe;
rs2::pipeline_profile profile = pipe.start();
auto dev = profile.get_device();
dev.hardware_reset();

RealSense-Customer-Engineering, 2018-08-28:
[Realsense Customer Engineering Team Comment]
Hi @fiorano10,

Did you try the workaround and see how it works? You can also add the reset in RealSenseNodeFactory::onInit() as below link.
https://github.com/ethz-asl/realsense/blob/devel/d400_node/realsense_ros_camera/src/realsense_node_factory.cpp

fiorano10, 2018-08-28:
I anyway have to reset the cameras using the realsense reset node to avoid the ‘Empty Frame’ error, so the hardware didn’t help. Does the reconfigure warning go away with the RealsenseNodeFactory reset?

RealSense-Customer-Engineering, 2018-08-28:
[Realsense Customer Engineering Team Comment]
Hi @fiorano10,

Please have a try of firmware 5.10.3 to see if it's fixed. So far, I don't see this input/output error on my test configuration when running dynamic reconfigure to toggle depth enable auto-exposure.

fiorano10, 2018-08-28:
@RealSense-Customer-Engineering I no longer see the input/output warning on a different machine after updating the FW and SDK.

closed 2018-08-28 by fiorano10