realsenseai/librealsense #2044: D435 infrared rectification issues

timprepscius, 2018-07-12:
D435 all platforms

--

I assume the rectified Y8 stereo images are calculated on the D4.

Is there a way to control the rectification of the stereo images?  They are losing quite a lot of quality.

I am seeing gaussian averaging, almost as if the D4 is reducing the image size, calculating rectification, and then increasing the image size again.  It is losing a lot of detail, and, it is losing the detail I need.

I can change the output Y16 and then apply rectification on my side.  But the Y16 seems to come in at FPS 25, which is very, very slow.

I really just want that Y16 at FPS 60, no rectification.  I don't need the depth image actually, just the stereo images.  Is this possible with a firmware upgrade?  Will this ever happen?

Is there any way you could open source the firmware? I would like to change it.

MartyG-RealSense, 2018-07-15:
The RealSense documentation states that Y16 is unrectified.  "Left and right infrared images are rectified by default (Y16 format is not)"

This is backed up by a statement by the Intel staff member 'RealSense Customer Engineering Team on this Issues forum': "The Y16 image is unrectified image and Y8 image is rectified image".

The combination of resolutions and FPS that will work with Y16 is limited.  One of these combinations is 1920x1080 Infrared 1 at 25 FPS and with Depth not enabled.  This confirms why your stream was coming in at 25 FPS.

[image]

The other resolution that works in this mode is 960x540.

It is unlikely the firmware would ever be open-sourced

D4 performs rectification of raw camera data received from the camera, yes, and it also performs alignment compensation..

timprepscius, 2018-07-19:
Hello,

I'm wondering what the probability is that Intel will change the firmware of the D4, to allow a passthrough of the 10bit infrared images, unaltered (un rectified, no gain, etc (or minimum gain, since there must be a reason for that 16.0)).  Or actually if they are rectified, fine, but still 10bit.

I need infrared images with more bandwidth in the contrast.
I would like some reasonable resolution, like 1280x800 at 90 fps.

The depths calculated from lower resolutions 848x480 are too unstable due to single pixel changes.
I don't need the D4 depth stream.

Within the bandwidth constraints of USB 3.1 this is possible.  And it seems like this should not be a huge deal ( Yes I know the irony of this statement and can hear the groans ).

Can I get some official statement like, "we are going to do this by quarter/year" or "this will never happen, sorry."

As dumb as this sounds, I am going to end up using a FPGA just to get to those images.

> 8.4
> Hardware met Software on the road to Changtse. Software said: "You are Yin and I am Yang. If we travel together, we will become famous and earn vast sums of money." And so they set forth together, thinking to conquer the world.
> 
> Presently, they met Firmware, who was dressed in tattered rags and hobbled along propped on a thorny stick. Firmware said to them: "The Tao lies beyond Yin and Yang. It is silent and still as a pool of water. It does not seek fame; therefore, nobody knows its presence. It does not seek fortune, for it is complete within itself. It exists beyond space and time."
> 
> Software and Hardware, ashamed, returned to their homes.

agrunnet, 2018-07-20:
Currently no plans for it.

timprepscius, 2018-07-20:
Ok, thanks for the information

closed 2018-07-26 by dorodnic