realsenseai/librealsense #2045: Working with a long USB 3.0 cable

Litvak1, 2018-07-12:
Required Info |  
-- | --
Camera Model | D415
Firmware Version | 05.08.15.00
Operating System & Version | Linux (Ubuntu 16.04)
Kernel Version (Linux Only) | 4.13.0-45-generic
Platform | PC
SDK Version | ?
Language | python

### Issue Description
Hi,

My D415 camera is attached to a manipulator arm robot, so I had to use a 5 meter cable to connect the camera to my PC. I'm using two USB 3.0 cables of 3m and 2m (in addition to the the small cable that came with the camera).

Basically, sometimes the camera works and sometimes it stops communicating. Every several minutes I need to unplug the cable and plug it back to fix the communication.

When I use realsense-viewer, after some time I get these error messages:

- Incomplete frame arrived
- xioctl(VIDIOC_DQBUF) failed last error
- USB REC overflow
- USB SCP overflow
- IR stream start failure
- Depth stream start failure

I work with Python, and I tried to reduce the size of the data from the camera by disabling the streams that I'm not using (I only use the depth stream in the highest resolution) and increasing the waiting time for a new frame to arrive. Here's the code:
```
    config = rs.config()
    config.disable_all_streams()
    config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
    config.enable_stream(rs.stream.infrared, 1280, 720, rs.format.y8, 30)
    pipeline = rs.pipeline()
    pipeline.start(config)   
    waiting_for_frame = True
    timeout_ms = 5000
    while waiting_for_frame:
        try:
            frames = pipeline.wait_for_frames(timeout_ms)
            waiting_for_frame = False
        except:
            pipeline.stop()
            timeout_ms += 1000
            time.sleep(1)
            pipeline.start(config)
```
What I did may improved the situation, but didn't solve the problem.

I know that using a 5 meter cable is not in the camera's specs, but this is a constraint that I have to work with.

Does anyone have a better solution for making the camera send depth images without crashing every several minutes?

Thanks

MartyG-RealSense, 2018-07-12:
Hi Litvak1,

The cables supplied by Newnex have been shown to work with the 400 Series cameras.

[https://communities.intel.com/message/533825#533825](url)

Litvak1, 2018-07-12:
Ho @MartyG-RealSense ,

Thank you for the fast reply! 

The 5m Newnex cable cost 50$ + 65$ shipment to my country. 

I prefer a programmable solution if there is one.

MartyG-RealSense, 2018-07-12:
I think you were on the right track with reducing the volume of data being sent through the cable.  You are likely to have less problems with the USB if you use a lower resolution and / or FPS rate, because this will reduce the amount of data bandwidth being sent through the cable.  For example, 848x400 is considered to be a resolution that can still provide good results.

Litvak1, 2018-07-12:
I have to work with 1280x720 resolution.

Do you think reducing the FPS will help?

I'll try that.

MartyG-RealSense, 2018-07-12:
Yes, reducing FPS should help.

Litvak1, 2018-07-12:
I reduced the FPS to 15. Lets see if the errors will disappear. 

If you have some time, I would appreciate if you could go over this issue as well (I have new information there since the last time we talked about it):
[Transforming pixel from a depth image to world coordinates](https://github.com/IntelRealSense/librealsense/issues/1904)

Thanks again!

RealSense-Customer-Engineering, 2018-07-19:
[Realsense Customer Engineering Team Comment]
hi @Litvak1,

how it works after set FPS to 15?
do you need more support on this?

Litvak1, 2018-07-21:
Hi,

It didn't fix the problem completely, but it did reduce the number of errors significantly!

Thanks for the tip!

I'll close this issue.

closed 2018-07-21 by Litvak1

CaleMcCollough, 2019-09-21:
This seems to me to be a power issue. We need to try a USB 3 cable advertised for fast charging your phone. I thought my unit was defective because I had 3 different cables, one the one that came with it, and plugging it into my desktop's front USB 3 port would cause my USB 3 card to go out. The Device Status said it malfunctioned and I would have to reset the computer. I plugged it into a USB 3 hub with an external power supply and used the cable that came with the real sense and it seems to be working, or at least a lot better than before. In OBS, it was freezing with Buffering enabled.

MartyG-RealSense, 2019-09-21:
@CaleMcCollough   I would suggest avoiding low-cost fast charging cables, as they will likely not be suitable for carrying the large amounts of data bandwidth that RealSense cameras transmit.

CaleMcCollough, 2019-09-22:
I have verified that it's, not the USB cable, it's the USB chipset. I have an Intel i5 laptop (Asus T300LA) and my cheap long USB cable works fine on the laptop but not on my  Gigabyte ga-970a-ud3p Rev 1 motherboard. I've tried to use a USB 3 hub with 2 one USB3 cable for data/power and another USB 2 for power and it still caused my internal USB 3 chipset to malfunction. Using another USB 3 hub with an external power supply, it still causes the USB Cam Overflow error from #2612. The higher quality cable only has slightly lower impedance, but it's still not in spec. Another problem is that I couldn't even update the firmware on my AMD desktop, it uploaded but it never showed up as completed.

MartyG-RealSense, 2019-09-23:
The 400 Series cameras are most likely to work on Intel or ARM processors.  There have been a couple of reports of them working with AMD Threadripper, but it is difficult from these limited reports to tell how well the cameras work with AMD.