realsenseai/librealsense #2047: DllNotFoundException: realsense2

arturdundur, 2018-07-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 2.13.0 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | RealSenseSDK2.0 | 
| Language                            | Unity 2018.1.0f2 (64bit) | 

Hello,

I'm trying to launch any example scene from 
`librealsense/wrappers/unity/Assets/RealSenseSDK2.0/Scenes/`
but always after i press play, console returns `DllNotFoundException: realsense2`.

I've followed guide providen in `librealsense/wrappers/unity/readme.md` 
and also tried copy files from `librealsense/wrappers/unity/Assets/RealSenseSDK2.0/` to unity  project,
in both cases the same error.

MartyG-RealSense, 2018-07-12:
In the SDK folder, you can find pre-made versions of the DLLs in the **Intel RealSense SDK 2.0 > bin > x64** folder.  In a case recently, a user who was having trouble getting DLLs they had compiled in Visual Studio to work in Unity was able to get it working just by dragging and dropping the pre-made DLLs into the Unity plugins folder.  This avoided the complex DLL compilation process in the Unity wrapper's instructions.

arturdundur, 2018-07-12:
@MartyG-RealSense Thanks, now seems to be working.
But another issue occurred 

> The script 'RealSenseDeviceInspectorEditor' does not derive from MonoBehaviour, but has been added to Game Object 'RealsenseDevice'. Please remove the script from the Game Object to avoid crashes.

Although with this I can manage, topic can be closed.

closed 2018-07-12 by arturdundur