realsenseai/librealsense #2048: librealsense 2.12 - cmake does not find libusb 1.0

bioingpv, 2018-07-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {MacOS}  | 
| Platform                                 | PC |
| SDK Version                            |  { 2.12.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I followed the installation_osx.md instructions, but
cmake does not see my libusb installation:
Stefanos-MacBook-Pro-2:build Stefano$ cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode

-- Info: REALSENSE_VERSION_STRING=2.12.0
CMake Warning at CMakeLists.txt:576 (message):
  Using libuvc!

-- Checking for one of the modules 'usb'
CMake Error at /usr/local/Cellar/cmake/3.11.4/share/cmake/Modules/FindPkgConfig.cmake:640 (message):
  None of the required 'usb' found
Call Stack (most recent call first):
  CMakeLists.txt:594 (pkg_search_module)

CMake Error at CMakeLists.txt:600 (message):
  Failed to find libusb-1.0

-- Configuring incomplete, errors occurred!
See also "/Users/Stefano/Downloads/librealsense-2.12.0/build/CMakeFiles/CMakeOutput.log".

On the other hand brew sees the library as installed and ok.
 
Stefanos-MacBook-Pro-2:build Stefano$ brew install libusb pkg-config
Warning: libusb 1.0.22 is already installed and up-to-date
To reinstall 1.0.22, run `brew reinstall libusb`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`

Any suggestion, please?
Thanks

dorodnic, 2018-07-12:
Hi @bioingpv 
I'm not sure why find_libusb fails on your system, but you could try adding `-D USE_SYSTEM_LIBUSB=false` to CMake parameters to ignore installed version.

bioingpv, 2018-07-12:
Hi @dorodnic, that worked! Thanks!

closed 2018-07-13 by RealSense-Customer-Engineering