realsenseai/librealsense #2050: Make error when Build RealSense SDK Samples for Android OS

cexopdc, 2018-07-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-24-generic                                        | 
| Platform                                 | Android  |
| SDK Version                            |  2.0                          | 

### Issue Description
Hi, I am trying to using the D435 camera on a Moto Z2 android device. I am in the middle of the "Build RealSense SDK Samples for Android OS" tutorial. 

I closely followed each steps in the tutorial, but in Step 7, after I execute commands: 
```
mkdir build && cd build
cmake .. -DANDROID_ABI=arm64-v8a -DCMAKE_TOOLCHAIN_FILE=/home/devplus/Documents/android-ndk-r17b/build/cmake/android.toolchain.cmake -DFORCE_LIBUVC=TRUE
make
```
After the "make" command, there are 9 errors generated. 

```
17 warnings and 9 errors generated.
CMakeFiles/realsense2.dir/build.make:88: recipe for target 'CMakeFiles/realsense2.dir/src/pipeline.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/pipeline.cpp.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
I then tried re-do all the steps from the start, or change the "DANDROID_ABI" from "arm64-v8a -DCMAKE_TOOLCHAIN" to "armeabi-v7a", but all resulting in the same error message as shown above.

Please let me know if there's anything I can try to make this work.

Thanks in advance.

dorodnic, 2018-07-12:
Hi @cexopdc 

There is currently a known issue with compilation for NDK >= 17 and a solution was proposed in #1853

For now, we are using the following [build script](https://github.com/IntelRealSense/librealsense/blob/master/.travis.yml#L125-L131) in our CI to verify we don't accidentally break Android compilation:
```
wget https://dl.google.com/android/repository/android-ndk-r16b-linux-x86_64.zip;
unzip android-ndk-r16b-linux-x86_64.zip -d ./;
mkdir build_android && cd build_android;
cmake .. -DCMAKE_TOOLCHAIN_FILE=../android-ndk-r16b/build/cmake/android.toolchain.cmake;
make -j $CPU_NUM;
```

RealSense-Customer-Engineering, 2018-07-19:
[Realsense Customer Engineering Team Comment]
@cexopdc Is the error resolved? Please update.

RealSense-Customer-Engineering, 2018-07-30:
[Realsense Customer Engineering Team Comment]
@cexopdc Could you please update if the issue resolved? Thanks!

cexopdc, 2018-07-30:
Thanks. The issue is with compilation for NDK >= 17, and the solution proposed in #1853 works like a charm.

Thanks for the help.

closed 2018-07-30 by cexopdc

qscesz19970810, 2022-09-01:
In fact I have tried it,but that not the problem.

That my cmake command "cmake .. -DANDROID_ABI=arm64-V8a -DCMAKE_TOOLCHAIN_FILE=/home/mds1/CrossCompiler/android-ndk-r21-linux-x86_64/android-ndk-r21/build/cmake/android.toolchatn.cmake -DFORCE_RSUSB_BACKEND=ON -DANDROID_PLATFORM=androtd-29 -CHECK_FOR_UPDATES=OFF"

My devices is  Android 11 , i think the sdk or ndk version maybe is a propblem......anyway i will use gradle wrapperis to build realsense.aar.