realsenseai/librealsense #2053: Intrinsics do not change after RGB Calibration

mso-michael, 2018-07-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.9.14| 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  NA                                  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  Robot                          | 

### Issue Description
We are performing repeated calibration of a Real Sense D415 to check how camera intrinsics vary. Using a target and the supplied Calibration Tools, we found some variation in the Depth calibration parameters, but none in RGB. It seems that despite following the Calibration software's dedicated RGB calibration step, no change is made to the factory intrinsics.

Is this to be expected? How come?

Thanks,

Michael

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
Hi @mso-michael,

Which calibration tool do you use? Dynamic calibration or OEM calibration? And, what's its version?

mso-michael, 2018-07-17:
Hi Realsense,

We use the Dynamic Calibration software version 2.6.4.0. We'd love to get an understanding of why RGB intrinsics are not updated, thanks for your help.

Michael

RealSense-Customer-Engineering, 2018-07-18:
[Realsense Customer Engineering Team Comment]
Hi @mso-michael,

Did you use CustomRW tool to read the parameters? The dynamic calibration is used to optimize extrinsic parameters. Intrinsic parameters (ex: distortion, principal point, etc) are not dynamically calibrated. You can refer to the dynamic calibration user guide to get more detail.
Only OEM calibration for factory use can calibrate the intrinsic and extrinsic parameters both. You can also refer to our custom calibration white paper to have your own calibration.

jb455, 2018-07-18:
Hi @RealSense-Customer-Engineering, sorry to butt in but is this OEM calibration tool available for general use? Or do you have to buy that [$1500 chessboard](https://click.intel.com/realsense-d400-camera-oem-calibration.html) to get it?

mso-michael, 2018-07-18:
Hi @RealSense-Customer-Engineering,

We calibrate using the dynamic calibration and read the published camera parameters using rostopic. There is clearly variation of the intrinsic parameters for the depth cameras (IR) after each calibration. Not much, but the intrinsic focal length parameter changes by 0.1-0.5%. If the Dynamic Calibration tool never calibrates intrinsic parameters, why would we be seeing this variation on the depth cameras? The fact the RGB intrinsics stay constant would make sense given your reply, but then why would the depth intrinsics change?

Thanks for your help,

Michael

RealSense-Customer-Engineering, 2018-07-20:
[Realsense Customer Engineering Team Comment]
Hi @mso-michael,

It's weird to have this result from rostopic. I will further check this. Thanks.

@jb455,

Yes, the chart you point is correct. The OEM calibration software is also available for NDA customers.

jb455, 2018-07-20:
@RealSense-Customer-Engineering 
Thanks, we are NDA. Is there someone specific we can contact about this or is it easier to go through our usual contacts?

RealSense-Customer-Engineering, 2018-07-20:
[Realsense Customer Engineering Team Comment]
Hi @jb455,

Please go thru your usual contacts and file the ticket on the Zendesk. You will get the access of the OEM calibration software.

RealSense-Customer-Engineering, 2018-07-25:
[Realsense Customer Engineering Team Comment]
Hi @mso-michael,

Please refer to #1526 to have more info. IR/depth image@640x480 is rectified, only Y16-format image is unrectified raw image.
That is the format @1920x1080/Y16 used for D415 dynamic calibration and you can see no change of intrinsic parameters showing in dynamic calibration tool.

RealSense-Customer-Engineering, 2018-08-02:
[Realsense Customer Engineering Team Comment]
Hi mso-michael,

still need any support for this topic?

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
Hi @mso-michael,

I will close this issue first. If any further question, you still can re-open or create another ticket.

closed 2018-08-09 by RealSense-Customer-Engineering