realsenseai/librealsense #2057: Realsense Viewer App instability while enabling and disabling sensor streams

napurvey, 2018-07-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version | MacOS High Sierra 10.13.4 (17E199) |                                 | 
| Platform                                 | Macbook Air |
| SDK Version                            |  v2.13 Development branch + Master                          | 
| Language                            |  C                         | 
| Segment			|  VR/AR                          | 

We have been trying to diagnose instability problems with our own Realsense capture app on Mac OSX, and as a test we have been running the Realsense viewer app to isolate the problem further.

After running the Realsense viewer app we have discovered that the same instabilities exist there. When first starting the app, selecting the stream configuration options and starting the color and depth streams, the following problems occur:

1. sometimes one of the Depth or Color streams will not even start resulting in the following:

[image]

2. sometimes when selecting streams you get this error:

[image]

3. most of the time after selecting and deselecting streams around ~6 times the sensor gets into a bad state and can lead to the Viewer app locking up (we have seen this happen in our own app):

[image]

The call stack when pausing in the Xcode debugger is like so:

[image]

We have the same symptoms in our own app, and therefore Realsense is not stable enough for us to ship due to instabilities when switching streams so would appreciate any fixes that address these issues. In this case I have been testing with the D435 but this also occurs on Mac OSX using a D415 sensor.

We have been previously using the master branch in our application but for these tests we have made sure we are up-to-date with the latest firmware and SDK and checked out to the Development branch just to soak up all the new fixes that have gone in, some of the solved bugs were also crashing our application.

Thanks!

MartyG-RealSense, 2018-07-13:
Collectively, the range of problems that you reported (frame timeouts, streams unavailable, etc) would make me suspect a stability issue with the USB port on your Mac, rather than a software issue.  This is more commonly reported on PCs, though perhaps because the number of people using RealSense on PC is greater and so the number of problems reported is proportionately greater.  There were a small number of reported USB-related issues with Macs on the older SR300 model of camera though.

napurvey, 2018-07-13:
Hi Marty,

Thanks for the the reply.

We have tested this on more than one Mac OSX machine; we have tested on two Macbook Air's (one of which was delivered and setup last week fresh), and a 2014 Mac Pro. I would deduce that this is more an OSX problem than a particular USB port problem unless a lot of Mac USB ports are not adequate enough to run a Realsense sensor on? This would cause a lot of problems for shipping Realsense support on Mac if this is the case.

MartyG-RealSense, 2018-07-13:
As of SDK version 2.13.0, the main unresolved Known Issue with OSX at the moment is that example programs written in the C language can freeze.  I guess you are probably not writing your app in C?  (I mean the old C, not C++ or C#)

A couple of other Known Issues currently relating to the firmware that may fit your symptoms are:

- Sporadic errors, only workaround is to physically reconnect camera 
- Frames didn't arrive error - after improper shutdown

napurvey, 2018-07-13:
Thanks for that info Marty; my takeaway is it seems at the moment that both sensor firmware and USB driver issues are present on Mac currently. I hope they become more stable in the future so we can confidently ship with this range of sensors.

RealSense-Customer-Engineering, 2018-07-19:
[Realsense Customer Engineering Team Comment]
hi @napurvey,

would like to know if you are NDA customer to intel? if yes, would suggest you to file ticket on Zendesk for direct support.

there are multiple issues here:
issue 1), i think i can reproduce this issue by below steps, please help to confirm it's same as you
1. connect D435 and start realsense viewer
2. open stereo module control and uncheck "Infrared 1"
3. start stereo module stream, no frame received for depth steam.

issue 2), seems like frame time out issue, when you stop and start streaming quickly, looks the previous frames not fully released and new stream start, will look into this issue. and from my test, if we stop/start slowly (after messages in the terminal stopped) then it is no problem, may be your short term workaround.
please also confirm repo steps:
1. connect D435 and start realsense viewer
2. start /stop depth stream quickly.

issue 3), Viewer app locking up issue, can you please provide more detail reproduce steps?
is this happened right after issue 2? do you unplug and plug the camera or after restart the viewer?

thanks.

RealSense-Customer-Engineering, 2018-07-19:
[Realsense Customer Engineering Team Comment]
update repo step for 1):
1. connect D435 and start realsense viewer
2. start Depth + IR streaming
3. open stereo module control and uncheck "Infrared 1"
4. start stereo module stream, no frame received for depth steam.

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
hi @napurvey,

thanks for the repo steps and yes, please file to Zendesk, thanks.

dorodnic, 2018-09-03:
Sorry for late reply, number 1 seem to be resolved on the development branch (according to the reproduction steps by @RealSense-Customer-Engineering ) We are still working on the rest.

dorodnic, 2018-09-25:
All the issues reported here should be resolved in [v2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1)

RealSense-Customer-Engineering, 2018-10-09:
[Realsense Customer Engineering Team Comment]
Hi @napurvey,

Did you try librs 2.16.1 to see how it works?

napurvey, 2018-10-09:
Hi,

Sorry, we are not in a position to test this currently due to higher
priorities driven by product, however this could change in the near future.

Thanks for the attention and support.

Neil

dorodnic, 2018-10-11:
I will close this issue for now. 
If you encounter any problem feel free to open new issues and / or contact support.

closed 2018-10-11 by dorodnic