realsenseai/librealsense #2058: D415 Pixel format hash requires patch and OpCodes do not match

SteveMacenski, 2018-07-14:
| Camera Model                       |D400  | 
| Firmware Version                   | 5.9.13 and 5.9.11 | 
| Operating System & Version |   (Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.14.13                                    | 
| Platform                                 | Intel NUC 7th gen i5, lenovo x1 carbon as well  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C                         | 
| Segment			|  Robot              | 

### Issue Description

I have 1 camera unit that does this and was working beforehand without issue. I tried reflashing the firmware, updating it, reverting it, using different cables, in different orientations but no luck.

```
 13/07 18:07:14,118 WARNING [140660045391744] (backend-v4l2.cpp:1098) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 13/07 18:07:14,118 WARNING [140660045391744] (sensor.cpp:313) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1531530434.127995013]: color stream is enabled - width: 320, height: 180, fps: 6
 13/07 18:07:14,128 WARNING [140660045391744] (types.cpp:57) OpCodes do not match! Sent 21 but received -6!
```

MartyG-RealSense, 2018-07-14:
Resolving this OpCode error us usually a two-step process.

1.  Confirm that your USB cable is good.  Sometimes, Type-C cables have quality issues.

2.  If you have a good USB cable then the next recommended step is to perform a calibration of the camera, as the error may be springing from a corrupted "calibration table".  

The link below can be used to obtain the tool to calibrate your camera.

https://downloadcenter.intel.com/download/27415/Intel-RealSense-D400-Series-Dynamic-Calibration-Software-Tools

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
hi @SteveMacenski,

would like to know which application you used for checking? realsense-viewer? and which release of libRealsense you are using?
thanks.

SteveMacenski, 2018-07-16:
I was using ros realsense, but it is also experienced in the realsense-viewer as the same errors appearing. I am using librealsense 2.10.2 as it is the latest matching version for the ros-realsense wrapper.

RealSense-Customer-Engineering, 2018-07-20:
[Realsense Customer Engineering Team Comment]
hi @SteveMacenski,

i think you can use new librealsense + ROS wrapper, and suggest to use newer version in case of we are talking on some issue been fixed already, thanks.

SteveMacenski, 2018-07-20:
The latest release of ros realsense is 2.0.3 which I'm using. I've also filed a ticket in that repo a few months ago asking for an updated release as well. The master branch does not consistently work with the current librealsense versions.

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
Hi @SteveMacenski,

1. You can use ros wrapper 2.0.3 + librealsense 2.12.0 which should work fine.
2. Need to have a patch in order to resolve below format issue. Suggest to downgrade to kernel 4.13 and the latest DKMS package (librealsense2-dkms) is available.

 13/07 18:07:14,118 WARNING [140660045391744] (backend-v4l2.cpp:1098) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 13/07 18:07:14,118 WARNING [140660045391744] (sensor.cpp:313) Unregistered Media formats : [RW16]; Supported: []

SteveMacenski, 2018-07-26:
This isnt a problem with my setup, I have a dozen other cameras that work fine with the same computer. It's only this unit that has this behavior

SteveMacenski, 2018-08-01:
The dynamic calibration tool does not work

```Serial Number: 810512060271
Device Name: Intel RealSense 415
Device PID:  0AD3
FW version:  05.09.11.00

Calibration Type: Targeted
Laser Power: On
Auto Exposure: On
AE SetPoint: Auto
Resolution: 1280x720x30

Error: calibration tables on device are not supported, Please try again with latest Dynamic Calibrator software. 
If you are already using latest software, please re-calibrate the device with latest OEM or Techinician Calibration tools and try again.
```

SteveMacenski, 2018-08-01:
> Intel.Realsense.CustomRW -g

closed 2018-08-01 by SteveMacenski