realsenseai/librealsense #2061: Unable to build release for android

rabits, 2018-07-14:
-------------------------------------------------------------------------------
| Required Info               |                                                 |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.9.14.0 | 
| Operating System & Version | Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  4.15.0-24                                         | 
| Platform                                 | PC amd64  |
| SDK Version                            |  master with #1853                       | 
| Language                            |  C                          | 

### Issue Description
During build with release flag:
```
$ cmake .. -DANDROID_ABI=arm64-v8a -DCMAKE_TOOLCHAIN_FILE=~/local/android-ndk-r17b/build/cmake/android.toolchain.cmake -DCMAKE_BUILD_TYPE=Release
$ make -j 48
```
 I see the next issue:
```
...
[ 77%] Linking CXX shared library librealsense2.so
CMakeFiles/realsense2.dir/src/ds5/advanced_mode/rs_advanced_mode.cpp.o: In function `librealsense::get_string(rs2_rs400_visual_preset)':
/home/user/Projects/librealsense/src/ds5/advanced_mode/rs_advanced_mode.cpp:25: undefined reference to `librealsense::make_less_screamy(char const*)'
/home/user/Projects/librealsense/src/ds5/advanced_mode/rs_advanced_mode.cpp:28: undefined reference to `librealsense::make_less_screamy(char const*)'
/home/user/Projects/librealsense/src/ds5/advanced_mode/rs_advanced_mode.cpp:26: undefined reference to `librealsense::make_less_screamy(char const*)'
/home/user/Projects/librealsense/src/ds5/advanced_mode/rs_advanced_mode.cpp:27: undefined reference to `librealsense::make_less_screamy(char const*)'
/home/user/Projects/librealsense/src/ds5/advanced_mode/rs_advanced_mode.cpp:30: undefined reference to `librealsense::make_less_screamy(char const*)'
CMakeFiles/realsense2.dir/src/ds5/advanced_mode/rs_advanced_mode.cpp.o:/home/user/Projects/librealsense/src/ds5/advanced_mode/rs_advanced_mode.cpp:29: more undefined references to `librealsense::make_less_screamy(char const*)' follow
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
CMakeFiles/realsense2.dir/build.make:1165: recipe for target 'librealsense2.so' failed
make[2]: *** [librealsense2.so] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
zsh: exit 2     make -j 48
```
Clean build in one thread also leads to the same error. The same with `development` branch...

MartyG-RealSense, 2018-07-14:
There was another Android case with Error 2 two days ago.  A solution was proposed by the Intel staff member in this discussion.

https://github.com/IntelRealSense/librealsense/issues/2050

rabits, 2018-07-14:
Hi @MartyG-RealSense, as I described I already applied the fix proposed for #2050 - #1853, unfortunately if fixes only default debug build. Difference in my build is using `-DCMAKE_BUILD_TYPE=Release` option.

MartyG-RealSense, 2018-07-14:
I apologize, I am replying on an Android tablet and the browser hid the end of your launch instruction, so I did not see your Release condition. 

Researching the Release instruction, I found an example where two dots are put at the end of the instruction.  Might this be relevant?

$ cmake -DCMAKE_BUILD_TYPE=Release ..

rabits, 2018-07-14:
@MartyG-RealSense no worries, my bad - already changed description)
Unfortunately no, it's the same issue in both ways.

MartyG-RealSense, 2018-07-14:
I unfortunately was not able to find any further leads to indicate what may be causing your issue with Release.  One of the Intel support members who has greater Android experience can likely provide a better answer.  I apologize for the wait in the meantime.  Good luck!

AnnaRomanov, 2018-08-12:
@rabits Hi, the problem is supposed to be fixed by #2189. Please let us know if it works for you now and we'll close this issue.

rabits, 2018-08-12:
Thanks @AnnaRomanov, right now I can build master for android without issues.

closed 2018-08-12 by rabits