realsenseai/librealsense #2071: Undefined reference to symbol 'XConvertSelection'

iglew, 2018-07-16:
Hi,

I'm having trouble building on Ubuntu 16.04. I followed all the installation instructions, but I keep getting the following error as soon as it starts to build the rs-capture example:
/usr/bin/ld: /usr/local/lib/libglfw3.a(x11_window.c.o): undefined reference to symbol 'XConvertSelection'

Googling it brings up loads of references to mistakes in the CMake files, but seeing as the library seems to be passing its internal build tests, is there something I'm missing?

iglew, 2018-07-18:
Um, I've not been linking any libraries in the command line when I compile it, am I supposed to? I followed these installation instructions line by line: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
And the error comes in at the "sudo make install" part. I've also tried compiling with cmake, both from the command line directly, and using CLion, and I get the same error. Am I supposed to be linking libraries when I compile it? It's not listed anywhere, and I have it compiling on MacOS with no problems, without linking anything extra at compile time.

RealSense-Customer-Engineering, 2018-07-18:
[Realsense Customer Engineering Team Comment]
@iglew
I thought you install the pre-built package and develop the application with gcc compilation. I didn't encounter and have never heard the problem you mention when build from source code on Ubuntu 16.04. Can you try to build the source from clean machine again?

BTW, does your Ubuntu with x window? or you only use the terminal? if so, you should build with CMake configuratio -DBUILD_GRAPHICAL_EXAMPLES=false. Also make sure the libglfw3 installed correctly.

RealSense-Customer-Engineering, 2018-07-30:
[Realsense Customer Engineering Team Comment]
@iglew
Is question still existing?

closed 2018-08-07 by RealSense-Customer-Engineering

sidmangla, 2018-10-18:
I have getting the same error. I have Ubuntu 14.04 and kernel -3.10.40 on toradex apalis tk1. I have cmake: 3.12
G++- 7.3
Gcc -7.3
Glfw-3.1.2