realsenseai/librealsense #2073: Filters for post processing doesn't apply

baptiste-mnh, 2018-07-17:
Hi,
I've an issue when I'm trying to use the filters for the depth map.
I use the exemple rs-post-processing as exemple but it doesn't work in my code.

## My code 
### Here is the function to export the depth map
I also show the render in 2 windows and I got no difference between the two of them.
Comparing to the realsense viewer with the same parameters, the result is not the same at all.

```
#include <rs/Filters/DepthFilter.hpp>
#include "exporter/DepthExporter.hpp"
#include <opencv2/core.hpp>
#include <opencv2/imgcodecs.hpp>
#include <opencv2/highgui.hpp>

void DepthExporter::export_scan(Camera &camera, rs2::frameset &frames, int nCapture, int seq) {
    rs2::colorizer color_map;
    color_map.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED, 1.f);
    color_map.set_option(RS2_OPTION_VISUAL_PRESET, 0.f);
    color_map.set_option(RS2_OPTION_COLOR_SCHEME, 0.f); 

    auto depth = frames.get_depth_frame();

    int width_raw = depth.get_width();
    int height_raw = depth.get_height();
    cv::Mat dMat_raw(cv::Size(width_raw, height_raw), CV_8UC3, (void *) color_map.colorize(depth).get_data());

    cv::namedWindow( "Filter from "+ camera.getSerial(), cv::WINDOW_AUTOSIZE ); 
    cv::imshow("Filter from "+ camera.getSerial(), dMat_raw);

    depth = DepthFilter::filter_depth(depth);
    int width = depth.get_width();
    int height = depth.get_height();
    cv::Mat dMat(cv::Size(width, height), CV_8UC3, (void *) color_map.colorize(depth).get_data());

    cv::namedWindow( "Depth from "+ camera.getSerial(), cv::WINDOW_AUTOSIZE );
    cv::imshow("Depth from "+ camera.getSerial(), dMat);
    cv::waitKey(10);

    saveFrame(dMat, getFullFileName(nCapture, camera, seq));
}
```
### Here is the part where I apply the filters
```
#include <librealsense2/hpp/rs_processing.hpp>
#include "rs/Filters/DepthFilter.hpp"

rs2::frame DepthFilter::filter_depth(rs2::frame &depth) {

    rs2::temporal_filter temp_filter;
    temp_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.15f);
    temp_filter.set_option(RS2_OPTION_FILTER_SMOOTH_DELTA, 100.0f);

    rs2::spatial_filter spatial_filter;
    spatial_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.4f);
    spatial_filter.set_option(RS2_OPTION_FILTER_SMOOTH_DELTA, 20.0f);
    spatial_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 2.0f);

    depth = temp_filter.process(depth);
    depth = spatial_filter.process(depth);
    return depth;
}
```
## Output
[image]

## Infos
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.11.0 | 
| Operating System & Version |   Ubuntu 18 | 
| Platform                                 | PC |
| SDK Version                            |  2.13.0                    | 
| Language                            |  C++                    |

dorodnic, 2018-07-17:
Hi @baptiste-mnh 
I would strongly discourage from declaring filters in local function scope. 
You could try making them `static` but better solution would adding them to `DepthFilter` object members. 
Filters have state (especially temporal filter, but also the rest) so if you reset them with every frame you will not get the right result.

baptiste-mnh, 2018-07-17:
Thank you for your response, it works now with one camera.
So if I apply this code for multiple camera at the same time, should I declare one filter by camera?

This file is very useful for those who want to know the proper way to declare all filters : 
librealsense/unit-tests/unit-tests-post-processing.cpp

dorodnic, 2018-07-17:
@baptiste-mnh 
Yes, you would need filter per camera. In the future, we are planning to abstract all this away into `pipeline` class (so you'd only say via `config` what kind of post-processing you want), but for now you need to manage it in your application. 
Also, I'm not sure if you managed to find [Librealsense Post-Processing Filters.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md) - this document also can be quite useful.

baptiste-mnh, 2018-07-17:
Yes it could be really great! Yes y sow this documentation but some exemple are missing.
I was searching for the persistance of the temporal filter.

Thanks for your help!

closed 2018-07-17 by baptiste-mnh