realsenseai/librealsense #2074: checkerboard 8X7 calibration file

omerbrandis, 2018-07-17:
Hello,

I wish to perform camara calibration using the 8X7 checkerboard,
but i was unable to locate file 540x480_60mm.pdf.

according to "Intel ® RealSenseTM Depth Module D400 Series Custom Calibration" rev 1.1.0 ( jan 2018)
this file should be located in "CustomCalibrationSample.zip".

i have downloaded CalibrationTools-2.6.4.0-Setup.zip but it does not contain file 540x480_60mm.pdf.
i have also installed the Calibration Tools, but the "target" directory also does not contain this file.
( it contained file print-target-fixed-width.pdf which does not contain a checkerboard).

please advise,
Omer.

MartyG-RealSense, 2018-07-17:
The PDF that you are searching for seems to have been replaced in June 2018 by the PDF linked below.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/print-target-fixed-width.pdf

omerbrandis, 2018-07-18:
100%,
Thanks,
Omer.

closed 2018-07-18 by omerbrandis