realsenseai/librealsense #2075: JNI ERROR (app bug): accessed stale local reference 0x200019 (index 6 in a table of size 1)

jamescartercamer, 2018-07-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {ubuntu 16.04}  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | Rooted Tecno CamonCX |
| SDK Version                            |  { legacy / 2.12.0}                          | 
| NDK                            |              {android-ndk-r16b  }        | 
| SDK		| {26 }                          | 

I just tried to build android Realsense_app following steps [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidJavaApp.md) but the final app crashes and displays JNI ERROR in the logcat consol. The infrared of my D400 is on

07-17 12:30:45.793 28664-28775/com.example.realsense_app I/OpenGLRenderer: Initialized EGL, version 1.4
07-17 12:30:46.965 28664-28664/com.example.realsense_app W/art: Before Android 4.1, method int android.support.v7.widget.DropDownListView.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView
07-17 12:30:46.102 28664-28768/com.example.realsense_app E/art: JNI ERROR (app bug): accessed stale local reference 0x200019 (index 6 in a table of size 1)
07-17 12:30:47.132 28664-28768/com.example.realsense_app A/art: art/runtime/java_vm_ext.cc:410] JNI DETECTED ERROR IN APPLICATION: use of deleted local reference 0x200019

bilalbinrais, 2018-07-19:
exact same issue here. logcat pulled through wifi/adb

07-20 00:08:54.258 23038-23038/? E/Zygote: v2
07-20 00:08:54.260 23038-23038/? E/Zygote: accessInfo : 0
07-20 00:08:54.902 23038-23059/com.example.realsense_app E/art: JNI ERROR (app bug): accessed stale local reference 0x20001d (index 7 in a table of size 1)
07-20 00:08:54.915 23038-23059/com.example.realsense_app A/art: art/runtime/java_vm_ext.cc:470] JNI DETECTED ERROR IN APPLICATION: use of deleted local reference 0x20001d
07-20 00:08:54.916 23038-23059/com.example.realsense_app A/art: art/runtime/java_vm_ext.cc:470] "Thread-8" prio=5 tid=13 Runnable
    art/runtime/java_vm_ext.cc:470]   | group="main" sCount=0 dsCount=0 obj=0x12e6d8b0 self=0x7f7448a200
    art/runtime/java_vm_ext.cc:470]   | sysTid=23059 nice=0 cgrp=default sched=0/0 handle=0x7f743ff450
    art/runtime/java_vm_ext.cc:470]   | state=R schedstat=( 107534053 1239692 138 ) utm=9 stm=1 core=2 HZ=100
    art/runtime/java_vm_ext.cc:470]   | stack=0x7f74305000-0x7f74307000 stackSize=1005KB
    art/runtime/java_vm_ext.cc:470]   | held mutexes= "mutator lock"(shared held)
    art/runtime/java_vm_ext.cc:470]   native: #00 pc 0000000000478734  /system/lib64/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiP12BacktraceMapPKcPNS_9ArtMethodEPv+220)
    art/runtime/java_vm_ext.cc:470]   native: #01 pc 0000000000478730  /system/lib64/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiP12BacktraceMapPKcPNS_9ArtMethodEPv+216)
    art/runtime/java_vm_ext.cc:470]   native: #02 pc 000000000044cca4  /system/lib64/libart.so (_ZNK3art6Thread9DumpStackERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEEbP12BacktraceMap+472)
    art/runtime/java_vm_ext.cc:470]   native: #03 pc 00000000002edfd0  /system/lib64/libart.so (_ZN3art9JavaVMExt8JniAbortEPKcS2_+1128)
    art/runtime/java_vm_ext.cc:470]   native: #04 pc 00000000002ee7a8  /system/lib64/libart.so (_ZN3art9JavaVMExt9JniAbortFEPKcS2_z+184)
    art/runtime/java_vm_ext.cc:470]   native: #05 pc 0000000000451858  /system/lib64/libart.so (_ZNK3art6Thread13DecodeJObjectEP8_jobject+352)
    art/runtime/java_vm_ext.cc:470]   native: #06 pc 0000000000100fb8  /system/lib64/libart.so (_ZN3art11ScopedCheck13CheckInstanceERNS_18ScopedObjectAccessENS0_12InstanceKindEP8_jobjectb+196)
    art/runtime/java_vm_ext.cc:470]   native: #07 pc 00000000000ff1c8  /system/lib64/libart.so (_ZN3art11ScopedCheck5CheckERNS_18ScopedObjectAccessEbPKcPNS_12JniValueTypeE+1120)
    art/runtime/java_vm_ext.cc:470]   native: #08 pc 0000000000104adc  /system/lib64/libart.so (_ZN3art8CheckJNI13CheckCallArgsERNS_18ScopedObjectAccessERNS_11ScopedCheckEP7_JNIEnvP8_jobjectP7_jclassP10_jmethodIDNS_10InvokeTypeEPKNS_7VarArgsE+140)
    art/runtime/java_vm_ext.cc:470]   native: #09 pc 0000000000103d9c  /system/lib64/libart.so (_ZN3art8CheckJNI11CallMethodVEPKcP7_JNIEnvP8_jobjectP7_jclassP10_jmethodIDSt9__va_listNS_9Primitive4TypeENS_10InvokeTypeE+704)
    art/runtime/java_vm_ext.cc:470]   native: #10 pc 00000000000f8e98  /system/lib64/libart.so (_ZN3art8CheckJNI21CallStaticVoidMethodVEP7_JNIEnvP7_jclassP10_jmethodIDSt9__va_list+96)
    art/runtime/java_vm_ext.cc:470]   native: #11 pc 00000000000204f0  /data/app/com.example.realsense_app-2/lib/arm64/libnative-lib.so (_ZN7_JNIEnv20CallStaticVoidMethodEP7_jclassP10_jmethodIDz+216)
    art/runtime/java_vm_ext.cc:470]   native: #12 pc 000000000001f7bc  /data/app/com.example.realsense_app-2/lib/arm64/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:470]   native: #13 pc 000000000001f5c0  /data/app/com.example.realsense_app-2/lib/arm64/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:470]   native: #14 pc 000000000001f590  /data/app/com.example.realsense_app-2/lib/arm64/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:470]   native: #15 pc 000000000001f37c  /data/app/com.example.realsense_app-2/lib/arm64/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:470]   native: #16 pc 000000000002da64  /data/app/com.example.realsense_app-2/lib/arm64/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:470]   native: #17 pc 000000000006885c  /system/lib64/libc.so (_ZL15__pthread_startPv+196)
    art/runtime/java_vm_ext.cc:470]   native: #18 pc 000000000001e118  /system/lib64/libc.so (__start_thread+16)
    art/runtime/java_vm_ext.cc:470]   (no managed stack frames)
    art/runtime/java_vm_ext.cc:470] 
07-20 00:08:54.957 23038-23059/com.example.realsense_app A/art: art/runtime/runtime.cc:419] Runtime aborting...
    art/runtime/runtime.cc:419] Aborting thread:
    art/runtime/runtime.cc:419] "Thread-8" prio=5 tid=13 Native
    art/runtime/runtime.cc:419]   | group="" sCount=0 dsCount=0 obj=0x12e6d8b0 self=0x7f7448a200
    art/runtime/runtime.cc:419]   | sysTid=23059 nice=0 cgrp=default sched=0/0 handle=0x7f743ff450
    art/runtime/runtime.cc:419]   | state=R schedstat=( 117342907 1379119 139 ) utm=9 stm=2 core=2 HZ=100
    art/runtime/runtime.cc:419]   | stack=0x7f74305000-0x7f74307000 stackSize=1005KB
    art/runtime/runtime.cc:419]   | held mutexes= "abort lock"
    art/runtime/runtime.cc:419]   native: #00 pc 0000000000478734  /system/lib64/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiP12BacktraceMapPKcPNS_9ArtMethodEPv+220)
    art/runtime/runtime.cc:419]   native: #01 pc 0000000000478730  /system/lib64/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiP12BacktraceMapPKcPNS_9ArtMethodEPv+216)
[... cut ...]
07-20 00:16:28.704 1589-1633/? E/BatteryStatsService: power: Missing API
07-20 00:16:28.713 1589-1633/? E/BatteryStatsService: no controller energy info supplied
07-20 00:16:28.714 2039-2039/? E/PhoneInterfaceManager: [PhoneIntfMgr] queryModemActivityInfo: Empty response
07-20 00:16:28.730 1589-1633/? E/BatteryStatsService: modem info is invalid: ModemActivityInfo{ mTimestamp=0 mSleepTimeMs=0 mIdleTimeMs=0 mTxTimeMs[]=[0, 0, 0, 0, 0] mRxTimeMs=0 mEnergyUsed=0}
07-20 00:16:32.862 1589-2011/? E/Watchdog: !@Sync 605 [20_Jul_00_16_32.861]

aarun2211, 2018-07-25:
I had a similar issue, and I was able to rectify it by converting the handlerClass in native-lib.cpp to a Global Ref. Please see [here](https://developer.android.com/training/articles/perf-jni) under the local and global ref section.

bilalbinrais, 2018-07-25:
@aarun2211 kudos. that solved my issue too.
I replaced the following line
`jclass handlerClass = env->FindClass(nativeClassName);`

with 
```
jclass localClass = env->FindClass(nativeClassName);
jclass handlerClass = reinterpret_cast<jclass>(env->NewGlobalRef(localClass));

```
To help others, im using NDK16b, realsense 2.12.0. SM-T820 device. Rooted with TWRP and SELinux set to permissive.

Additionally changes in the following pull were made too.

https://github.com/IntelRealSense/librealsense/pull/1853/commits/613722a3e6b4b0eb0bb6e839644bcd1fd0b67b71

aarun2211, 2018-07-25:
glad it worked out!!

closed 2018-07-31 by RealSense-Customer-Engineering