realsenseai/librealsense #2081: Image of depth sensor inwindow inside a VB.NET Form

Maanbalo, 2018-07-17:
| Camera Model D435
| Firmware Version 05.09.13.00
| Operating System & Version Win 10 Pro for OEM
| Platform PC
| SDK Version .NET 4.6
| Language VB.NET
| Segment	Others

I am developing an app for my D435 and I am doing it in VB.NET, 

How can I show the image of an infrared sensor or depth sensor in a box window inside a VB.NET form?

valkjsaaa, 2018-07-17:
I think you should be able to use the C# library in VB .net.

RealSense-Customer-Engineering, 2018-07-18:
[Realsense Customer Engineering Team Comment]
yes, as valkjsaaa said, use the C# library for VB.net development.

Maanbalo, 2018-07-19:
Yes, I use Intel.Realsense.dll but i don't know the code for transform the data sensor to picture for show it  in a PictureBox

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
@Maanbalo
I am really not familiar with the VB.NET and tried to write an sample code to you, the code may be messy. I only wrote the sample code to display RGB image, please write the code by yourself for the colorized depth image.
`​Imports Intel.RealSense`
`Imports System.ComponentModel`
`Imports System.Runtime.InteropServices`
`Imports System.Windows.Threading`

`Public Class Form1`
`    Dim pipeline As New Pipeline()`
`    Dim coloriser As New Colorizer()`
`    Dim cfg As New Config()`
`    Dim th As System.Threading.Thread = New Threading.Thread(AddressOf Retrieve_Frames)`
`    Dim stopThread As Boolean`
`    Public Sub Form1_Load(sender As Object, e As EventArgs) Handles MyBase.Load`
`        stopThread = False`
`        cfg.EnableStream(Intel.RealSense.Stream.Depth, 640, 480, Format.Z16, 30)`
`        cfg.EnableStream(Intel.RealSense.Stream.Color, 640, 480, Format.Bgr8, 30)`
`        Dim pipelineprofile As PipelineProfile = pipeline.Start(cfg)`
`        th.Start()`
`    End Sub`

`    Public Sub Form1_Unload() Handles MyBase.FormClosing`
`        stopThread = True`
`        th.Join()`
`        pipeline.Stop()`
`    End Sub`

`    Public Function UploadImage(pic As PictureBox, frame As VideoFrame)`
`        Dim bytes(frame.Stride * frame.Height - 1) As Byte`
`        frame.CopyTo(Of Byte)(bytes)`
`        Dim bitmap As Bitmap = New Bitmap(frame.Width, frame.Height, frame.Stride, System.Drawing.Imaging.PixelFormat.Format24bppRgb, Marshal.UnsafeAddrOfPinnedArrayElement(bytes, 0))`
`        pic.Image = bitmap`
`    End Function`

`    Public Function Retrieve_Frames()`
`        Do While stopThread <> True`
`            Dim frames = pipeline.WaitForFrames()`
`            If (frames.Count < 2) Then`
`                Continue Do`
`            End If`
`            Dim depth = frames.DepthFrame`
`            Dim color = frames.ColorFrame`
`            Dim videoFrame As VideoFrame = TryCast(color, VideoFrame)`
`            UploadImage(PictureBox1, videoFrame)`
`        Loop`

`    End Function`

`End Class`

``

Maanbalo, 2018-07-23:
Hi, the code is correct. Thank you. XD

closed 2018-07-23 by Maanbalo