realsenseai/librealsense #2083: Missing release .deb or Arch Linux Package for librealsense 1.9.7

Kuchiriel, 2018-07-18:
And when I try to compile from source I get this error.
`λ trevor [~/librealsense-1.9.7] → make
g++ src/hw-monitor.cpp -std=c++11 -fPIC -pedantic -Ofast -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -mssse3 -c -o obj/hw-monitor.o
In file included from src/uvc.h:8,
                 from src/hw-monitor.h:8,
                 from src/hw-monitor.cpp:5:
src/types.h:572:14: error: ‘function’ in namespace ‘std’ does not name a template type
         std::function<void()> continuation;
              ^~~~~~~~
src/types.h:572:9: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
src/types.h:25:1:
+#include <functional>
 
src/types.h:572:9:
         std::function<void()> continuation;
         ^~~
src/types.h:580:50: error: expected ‘)’ before ‘<’ token
         explicit frame_continuation(std::function<void()> continuation, const void* protected_data) : continuation(continuation), protected_data(protected_data) {}
                                    ~             ^
                                                  )
src/types.h: In constructor ‘rsimpl::frame_continuation::frame_continuation()’:
src/types.h:578:32: error: class ‘rsimpl::frame_continuation’ does not have any field named ‘continuation’
         frame_continuation() : continuation([]() {}) {}
                                ^~~~~~~~~~~~
src/types.h: In constructor ‘rsimpl::frame_continuation::frame_continuation(rsimpl::frame_continuation&&)’:
src/types.h:583:59: error: class ‘rsimpl::frame_continuation’ does not have any field named ‘continuation’
         frame_continuation(frame_continuation && other) : continuation(std::move(other.continuation)), protected_data(other.protected_data)
                                                           ^~~~~~~~~~~~
src/types.h:583:88: error: ‘class rsimpl::frame_continuation’ has no member named ‘continuation’; did you mean ‘frame_continuation’?
 nuation(frame_continuation && other) : continuation(std::move(other.continuation)), protected_data(other.protected_data)
                                                                     ^~~~~~~~~~~~
                                                                     frame_continuation
src/types.h:585:19: error: ‘class rsimpl::frame_continuation’ has no member named ‘continuation’; did you mean ‘frame_continuation’?
             other.continuation = []() {};
                   ^~~~~~~~~~~~
                   frame_continuation
src/types.h: In member function ‘void rsimpl::frame_continuation::operator()()’:
src/types.h:591:13: error: ‘continuation’ was not declared in this scope
             continuation();
             ^~~~~~~~~~~~
src/types.h:591:13: note: suggested alternative: ‘frame_continuation’
             continuation();
             ^~~~~~~~~~~~
             frame_continuation
src/types.h: In member function ‘void rsimpl::frame_continuation::reset()’:
src/types.h:599:13: error: ‘continuation’ was not declared in this scope
             continuation = [](){};
             ^~~~~~~~~~~~
src/types.h:599:13: note: suggested alternative: ‘frame_continuation’
             continuation = [](){};
             ^~~~~~~~~~~~
             frame_continuation
src/types.h: In member function ‘rsimpl::frame_continuation& rsimpl::frame_continuation::operator=(rsimpl::frame_continuation&&)’:
src/types.h:606:13: error: ‘continuation’ was not declared in this scope
             continuation();
             ^~~~~~~~~~~~
src/types.h:606:13: note: suggested alternative: ‘frame_continuation’
             continuation();
             ^~~~~~~~~~~~
             frame_continuation
src/types.h:608:34: error: ‘class rsimpl::frame_continuation’ has no member named ‘continuation’; did you mean ‘frame_continuation’?
             continuation = other.continuation;
                                  ^~~~~~~~~~~~
                                  frame_continuation
src/types.h:609:19: error: ‘class rsimpl::frame_continuation’ has no member named ‘continuation’; did you mean ‘frame_continuation’?
             other.continuation = []() {};
                   ^~~~~~~~~~~~
                   frame_continuation
src/types.h: In destructor ‘rsimpl::frame_continuation::~frame_continuation()’:
src/types.h:616:13: error: ‘continuation’ was not declared in this scope
             continuation();
             ^~~~~~~~~~~~
src/types.h:616:13: note: suggested alternative: ‘frame_continuation’
             continuation();
             ^~~~~~~~~~~~
             frame_continuation
src/types.h: At global scope:
src/types.h:625:44: error: expected ‘)’ before ‘<’ token
         calibration_validator(std::function<bool(rs_stream, rs_stream)> extrinsic_validator,
                              ~             ^
                                            )
src/types.h:633:14: error: ‘function’ in namespace ‘std’ does not name a template type
         std::function<bool(rs_stream from_stream, rs_stream to_stream)> extrinsic_validator;
              ^~~~~~~~
src/types.h:633:9: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
         std::function<bool(rs_stream from_stream, rs_stream to_stream)> extrinsic_validator;
         ^~~
src/types.h:634:14: error: ‘function’ in namespace ‘std’ does not name a template type
         std::function<bool(rs_stream stream)> intrinsic_validator;
              ^~~~~~~~
src/types.h:634:9: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
         std::function<bool(rs_stream stream)> intrinsic_validator;
         ^~~
make: *** [Makefile:106: obj/hw-monitor.o] Error 1
`

dorodnic, 2018-07-18:
Hi @Kuchiriel 
Why are you installing 1.9.7 specifically, and not one of the later legacy releases (like [1.11.0](https://github.com/IntelRealSense/librealsense/releases/tag/stable))?

dorodnic, 2018-07-18:
If this is the only error, I'd try adding `#include <functional>` to `hw-monitor.cpp` before any other line. 
Not sure why it would compile with Ubuntu but not Arch...

Kuchiriel, 2018-07-20:
Dit it. 

 make                  
g++ src/hw-monitor.cpp -std=c++11 -fPIC -pedantic -Ofast -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0`  -mssse3 -c -o obj/hw-monitor.o
In file included from src/uvc.h:8,
                 from src/hw-monitor.h:8,
                 from src/hw-monitor.cpp:5:
src/types.h:572:14: error: ‘function’ in namespace ‘std’ does not name a template type
         std::function<void()> continuation;
              ^~~~~~~~
src/types.h:572:9: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
src/types.h:25:1:
+#include <functional>
 
src/types.h:572:9:
         std::function<void()> continuation;
         ^~~
src/types.h:580:50: error: expected ‘)’ before ‘<’ token
         explicit frame_continuation(std::function<void()> continuation, const void* protected_data) : continuation(continuation), protected_data(protected_data) {}
                                    ~             ^
                                                  )
src/types.h: In constructor ‘rsimpl::frame_continuation::frame_continuation()’:
src/types.h:578:32: error: class ‘rsimpl::frame_continuation’ does not have any field named ‘continuation’
         frame_continuation() : continuation([]() {}) {}
                                ^~~~~~~~~~~~
src/types.h: In constructor ‘rsimpl::frame_continuation::frame_continuation(rsimpl::frame_continuation&&)’:
src/types.h:583:59: error: class ‘rsimpl::frame_continuation’ does not have any field named ‘continuation’
         frame_continuation(frame_continuation && other) : continuation(std::move(other.continuation)), protected_data(other.protected_data)
                                                           ^~~~~~~~~~~~
src/types.h:583:88: error: ‘class rsimpl::frame_continuation’ has no member named ‘continuation’; did you mean ‘frame_continuation’?
 nuation(frame_continuation && other) : continuation(std::move(other.continuation)), protected_data(other.protected_data)
                                                                     ^~~~~~~~~~~~
                                                                     frame_continuation
src/types.h:585:19: error: ‘class rsimpl::frame_continuation’ has no member named ‘continuation’; did you mean ‘frame_continuation’?
             other.continuation = []() {};
                   ^~~~~~~~~~~~
                   frame_continuation
src/types.h: In member function ‘void rsimpl::frame_continuation::operator()()’:
src/types.h:591:13: error: ‘continuation’ was not declared in this scope
             continuation();
             ^~~~~~~~~~~~
src/types.h:591:13: note: suggested alternative: ‘frame_continuation’
             continuation();
             ^~~~~~~~~~~~
             frame_continuation
src/types.h: In member function ‘void rsimpl::frame_continuation::reset()’:
src/types.h:599:13: error: ‘continuation’ was not declared in this scope
             continuation = [](){};
             ^~~~~~~~~~~~
src/types.h:599:13: note: suggested alternative: ‘frame_continuation’
             continuation = [](){};
             ^~~~~~~~~~~~
             frame_continuation
src/types.h: In member function ‘rsimpl::frame_continuation& rsimpl::frame_continuation::operator=(rsimpl::frame_continuation&&)’:
src/types.h:606:13: error: ‘continuation’ was not declared in this scope
             continuation();
             ^~~~~~~~~~~~
src/types.h:606:13: note: suggested alternative: ‘frame_continuation’
             continuation();
             ^~~~~~~~~~~~
             frame_continuation
src/types.h:608:34: error: ‘class rsimpl::frame_continuation’ has no member named ‘continuation’; did you mean ‘frame_continuation’?
             continuation = other.continuation;
                                  ^~~~~~~~~~~~
                                  frame_continuation
src/types.h:609:19: error: ‘class rsimpl::frame_continuation’ has no member named ‘continuation’; did you mean ‘frame_continuation’?
             other.continuation = []() {};
                   ^~~~~~~~~~~~
                   frame_continuation
src/types.h: In destructor ‘rsimpl::frame_continuation::~frame_continuation()’:
src/types.h:616:13: error: ‘continuation’ was not declared in this scope
             continuation();
             ^~~~~~~~~~~~
src/types.h:616:13: note: suggested alternative: ‘frame_continuation’
             continuation();
             ^~~~~~~~~~~~
             frame_continuation
src/types.h: At global scope:
src/types.h:625:44: error: expected ‘)’ before ‘<’ token
         calibration_validator(std::function<bool(rs_stream, rs_stream)> extrinsic_validator,
                              ~             ^
                                            )
src/types.h:633:14: error: ‘function’ in namespace ‘std’ does not name a template type
         std::function<bool(rs_stream from_stream, rs_stream to_stream)> extrinsic_validator;
              ^~~~~~~~
src/types.h:633:9: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
         std::function<bool(rs_stream from_stream, rs_stream to_stream)> extrinsic_validator;
         ^~~
src/types.h:634:14: error: ‘function’ in namespace ‘std’ does not name a template type
         std::function<bool(rs_stream stream)> intrinsic_validator;
              ^~~~~~~~
src/types.h:634:9: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
         std::function<bool(rs_stream stream)> intrinsic_validator;
         ^~~

RealSense-Customer-Engineering, 2018-08-02:
[Realsense Customer Engineering Team Comment]
@Kuchiriel
Can you try to add the #include <functional> to the include section of "types.h"? you can refer to the latest version types.h, the functional include is there.

Kuchiriel, 2018-08-02:
@RealSense-Customer-Engineering First, thanks for supporting me in this. I didn't get it, what I need do exactly? Include  #include <functional>?
Thanks!

RealSense-Customer-Engineering, 2018-08-06:
[Realsense Customer Engineering Team Comment]
@Kuchiriel
I searched online, found out the error you encountered usually because of lacking the include of functional. And from the error message you attached, the error comes form types.h, so I guess you can add the "#include <functional>" into the types.h and see if it could fix the problem. Also in the latest version of types.h file, the functional is included in the header file.

RealSense-Customer-Engineering, 2018-08-14:
[Realsense Customer Engineering Team Comment]
Any update?

Kuchiriel, 2018-08-14:
@RealSense-Customer-Engineering 
```
Package 'glfw3', required by 'virtual:world', not found
examples/c-tutorial-3-pointcloud.c:20:10: fatal error: GLFW/glfw3.h: No such file or directory
 #include <GLFW/glfw3.h>
          ^~~~~~~~~~~~~~
compilation terminated.
```

Kuchiriel, 2018-08-17:
@RealSense-Customer-Engineering There is no package of libglfw3 for Manjaro Linux :/
Can I install manually?

dorodnic, 2018-08-19:
We provide the following script to build and install glfw from source: [scripts/install_glfw3.sh](https://github.com/IntelRealSense/librealsense/blob/master/scripts/install_glfw3.sh)

Kuchiriel, 2018-08-24:
Make went successfully:

`make            
cc examples/c-tutorial-3-pointcloud.c -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/c-tutorial-3-pointcloud
cc examples/c-tutorial-1-depth.c -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/c-tutorial-1-depth
cc examples/c-tutorial-2-streams.c -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/c-tutorial-2-streams
g++ examples/cpp-stride.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-stride
g++ examples/cpp-tutorial-3-pointcloud.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-tutorial-3-pointcloud
g++ examples/cpp-multicam.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-multicam
g++ examples/cpp-enumerate-devices.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-enumerate-devices
g++ examples/cpp-headless.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-headless
g++ examples/cpp-alignimages.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-alignimages
g++ examples/cpp-motion-module.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-motion-module
g++ examples/cpp-callback-2.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-callback-2
g++ examples/cpp-pointcloud.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-pointcloud
g++ examples/cpp-restart.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-restart
g++ examples/cpp-tutorial-1-depth.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-tutorial-1-depth
g++ examples/cpp-tutorial-2-streams.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-tutorial-2-streams
g++ examples/cpp-callback.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-callback
g++ examples/cpp-capture.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-capture
g++ examples/cpp-config-ui.cpp -std=c++11 -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/cpp-config-ui
g++ unit-tests/*.cpp -std=c++11 -o bin/tests/F200-live-test -DLIVE_TEST -DF200_TEST -DMAKEFILE -Iinclude -Llib -lrealsense -lm
g++ unit-tests/*.cpp -std=c++11 -o bin/tests/LR200-live-test -DLIVE_TEST -DLR200_TEST -DMAKEFILE -Iinclude -Llib -lrealsense -lm
g++ unit-tests/*.cpp -std=c++11 -o bin/tests/R200-live-test -DLIVE_TEST -DR200_TEST -DMAKEFILE -Iinclude -Llib -lrealsense -lm
g++ unit-tests/*.cpp -std=c++11 -o bin/tests/SR300-live-test -DLIVE_TEST -DSR300_TEST -DMAKEFILE -Iinclude -Llib -lrealsense -lm
g++ unit-tests/*.cpp -std=c++11 -o bin/tests/ZR300-live-test -DLIVE_TEST -DZR300_TEST -DMAKEFILE -Iinclude -Llib -lrealsense -lm
g++ unit-tests/*.cpp -std=c++11 -o bin/tests/offline-test -DOFFLINE_TEST -Doffline_TEST -DMAKEFILE -Iinclude -Llib -lrealsense -lm
`
Also make install:
`sudo make install
install -m755 -d /usr/local/include/librealsense
cp -r include/librealsense/* /usr/local/include/librealsense
cp lib/librealsense.so /usr/local/lib
ldconfig
`

RealSense-Customer-Engineering, 2018-09-18:
[Realsense Customer Engineering Team Comment]
Hi @kuchiriel,

Did you still have the issue?
Did you set the installation destination path in CMAKE configuration? or, set "DESTDIR" in make command?

Kuchiriel, 2018-09-27:
Hi @RealSense-Customer-Engineering, when I just use make, after making those modifications, it compiles sucessfully. But with Cmake I got those errors. I did not set the installation destination path in CMAKE configuration nor seted "DESTDIR" in make command. I'm kinda newbie with C and C++. I'm web developer, but my problem was solved, make works fine. I thank you all for the help! I managed to make pupil run on my system. When I have some time I will try to fix the Cmake issue, and resubmit to AUR.

RealSense-Customer-Engineering, 2018-09-27:
[Realsense Customer Engineering Team Comment]
Ok, if any update, let us know.