realsenseai/librealsense #2084: Error: "A device attached to the system is not functioning"

xtrawurst, 2018-07-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |  Win10  | 
| SDK Version                            |  2.14.0                      | 

### Issue Description
Every once in a while (about every 20th to 50th start of the sensor) I get an exception: "A device attached to the system is not functioning" (see attached error screenshot from the RS viewer).

Whenever I do get the error the code that's causing it (in our app) is
```
auto list = rs_ctx->query_devices();

if (list.size() == 0) {
  return NO_SENSOR_CONNECTED;
}

auto dev = list[0];  // <-- Throws here
```
[image]

MartyG-RealSense, 2018-07-18:
According to the SDK 2.0 beginners-guide documentation, an alternate way to write this camera detection function is:

*********

rs2::context ctx;
auto list = ctx.query_devices(); // Get a snapshot of currently connected devices

if (list.size() == 0) 
 throw std::runtime_error("No device detected. Is it plugged in?");

rs2::device dev = list.front();

xtrawurst, 2018-07-18:
@MartyG-RealSense 
list.front() or list[0]. It makes no difference.

MartyG-RealSense, 2018-07-18:
I compared the firmware notes for differences after firmware 5.9.11 (where you never have problems).  In 5.9.13, fixes were added to try to address the issue of the camera being non-detected after a system reboot in USB 3.0 or USB 2.0 mode, and the issue of the camera stalling after waking from sleep in USB 2.0 mode.

xtrawurst, 2018-07-18:
@MartyG-RealSense 

The camera is not running in USB2 mode.

MartyG-RealSense, 2018-07-18:
Thank you very much for the information.

At this point, the best course of action is  likely to be to wait for input from a member of the RealSense support team on this forum, such as 'RealSense-Customer-Engineering',  about their thoughts on what the cause may be.

xtrawurst, 2018-07-18:
I just tested with the laptop disconnected from power (i.e. running from battery) and the described issue happened two times in a row. Could this have something to do with USB power?

MartyG-RealSense, 2018-07-18:
I'm aware of a case involving another type of error often associated with USB ("No device found that satisfies all requirements") where the user's camera started working after they removed the batteries from their laptop.

dorodnic, 2018-07-18:
Hi @xtrawurst 
Unfortunately, `A device attached to the system is not functioning` is essentially the generic "something went wrong" error of Media Foundation. We observed this HResult many times in the past, more often on Windows 8 and 7. Now days, Microsoft are making changes to Media Foundation drivers in their rolling updates, so it is critical to know the exact Windows 10 build you are using.

jb455, 2018-07-30:
I've started getting this with a D410 on Win10 1803 build 17134.48, SDK 2.14.1 and FW 05.09.13&14

The first couple of times I start the camera up in my app it's fine then when I try again (could be the 2nd,3rd,5th... time) I get this error message then the next time after that, the camera has disappeared and a restart is required to get it working again. Downgrading to FW 05.09.11 appears to stop this issue, thanks @xtrawurst for providing that information!

RealSense-Customer-Engineering, 2018-08-01:
[Realsense Customer Engineering Team Comment]
hi @jb455,

would like to reproduce this issue so want to know if you use realsense viewer can reproduce this issue?
and steps?
just start video stream (depth only or? )
stop video stream and start again?
thanks.

jb455, 2018-08-01:
Hi,
I was using the depth stream in a software device in C#, not sure if that's relevant. I couldn't reproduce it just turning the stream off and on in realsense viewer, or opening and closing the app; but the error was coming up if I opened it up in my app a couple of times first then opened viewer.
The basic flow of my app is:

1. Query devices, select D410
2. Query streams, select desired depth stream
3. Set stream options (depth units, visual preset)
4. Use `Sensor.Open(StreamProfile)` and `Sensor.Start(FrameQueue)` to start the stream
5. `Frame = FrameQueue.WaitForFrame()` to get frames; copying data bytes & some other properties to pass to software sensor
6. `Dispose` frame when not needed any more in each loop
7. Call `StreamProfile.Dispose()`, `Sensor.Stop()`, `Sensor.Close()` when camera not needed any more.

Going through this whole process several times is what was causing the issue.

RealSense-Customer-Engineering, 2018-08-27:
[Realsense Customer Engineering Team Comment]
hi @jb455,

your process seems normal to me, so i try to reproduce this issue with  our csharp samples, try many times and can't reproduce this issue.
also try to unplug power to have PC run in battery mode, still no fail.
checked my win 10 is build 1709,  going to find other win10 system to have a try.

RealSense-Customer-Engineering, 2018-09-26:
[Realsense Customer Engineering Team Comment]
Hi @jb455,

Not see this issue with "cs-tutorial-1-depth" C# app on the latest librealsense + Win10 1803 build 10.0.17134.
Please verify it to see how it works.

closed 2018-11-09 by RealSense-Customer-Engineering