realsenseai/librealsense #2085: ERROR 'Failed to resolve request. No device found that satisfies all requirements'

vgenten, 2018-07-18:
I'm trying to start a stream of my D435 similar to https://github.com/IntelRealSense/librealsense/tree/master/examples/capture
I need to add a configuration to the pipeline like
cfg.enable_stream(RS2_STREAM_COLOR,2,1920,1080,RS2_FORMAT_BGR8,30)
to access the rgb stream with the given resolution. But running it gives me this error:
"terminate called after throwing an instance of 'rs2::error'
  what():  Failed to resolve request. No device found that satisfies all requirements"

What am I doing wrong?
Thanks for your help

AntoineWefit, 2018-07-18:
I don't own a D435 so I'm not sure but, what is this "2" right after "RS2_STREAM_COLOR" ?
I reviewed my own code and I don't use any parameter here, maybe you can try deleting it ? I could be wrong though, could you post a bigger part of your code ?

I would also suggest you to use RGB8 format instead of the default BGR8, unless you want blueish captures :)

MartyG-RealSense, 2018-07-18:
In most of the cases I have dealt with where the 'No device found that satisfies all requirements' error was reported, it was because of the USB-C cable that was being used with the camera.   Can you tell me please if you are using the Intel-supplied USB cable that was provided with the camera?

The one case where the USB cable was apparently not the cause was reported to involve the stream configuration.  The user said the following:

"I have found that in Python 2.7, config.enable_stream doesn't work properly.   For all except the color_image stream, the setting is highly particular... and only one setting works.   For example, if I set the IR to anything else (but 640x480), I get the  [No device found that satisfies all requirements'] error message".

Edit: Antoine has a point.  I would expect the color stream instruction to look like this:

cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 30);

vgenten, 2018-07-19:
Thanks for your reply. When I call it without the index and using RGB8 format I get a black screen.
Here's my code
    window app(1920,1080, "RealSense Capture Example");
    texture color_image;
    colorizer color_map;
    pipeline pipe;
    config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR,1920,1080,RS2_FORMAT_RGB8,30);
    auto profile = pipe.start(cfg);
    while (app) {
      frameset data = pipe.wait_for_frames();
      frame color = data.get_color_frame(); 
      color_image.render(color, { app.width(), 0, app.width(), app.height() });
    }

I use the included USB cable from Intel.

MartyG-RealSense, 2018-07-19:
Stream programming is not one of my specializations, so I apologize for gaps in my knowledge in that regard.  Comparing your script to the 'Capture' example that your script is based on, I notice that you are not using the rs2:: instruction at the start of your lines that this example script and other scripts use.  Do you think this could be significant?

https://github.com/IntelRealSense/librealsense/blob/master/examples/capture/readme.md

vgenten, 2018-07-19:
That is because I'm using a namespace.
It's working now if I convert it to opencv, maybe I'm using the render function not correctly?
Thanks for your help!

MartyG-RealSense, 2018-07-19:
Thanks for the information about namespace use.  I'm not an everyday RealSense coder.  I learn as I go.  :)

You're very welcome.  Great news that you got it working!

closed 2018-08-09 by RealSense-Customer-Engineering