realsenseai/librealsense #2087: internal processing hardware D435

pierref33, 2018-07-19:
Hello,

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | (05.09.11.00) | 
| Operating System & Version |   Win (10)   |                                     | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | opencv/python                         | 

### Issue Description
 Is the internal processing hardware of the cameraD435 (D4 ASIC) exclusively devoted to provide the best images and depth precision? Is there any ways to do some processing in the camera as an embedded system or all the high-level processing must be performed inevitably on the host computer ?

Thank you in advance !

MartyG-RealSense, 2018-07-19:
The Vision Processor D4 is designed to perform processes such as rectification to reduce the amount of calculations that the computing host-processor has to do.  The results of those calculations are then output over USB.  Some form of computing board should be required though to do something useful with that data output.

You would not necessarily have to connect the camera to a full PC.  System integrators could incorporate small 'single board' computers such as the Up Board and connect the camera to that.  Or if computing power closer to a desktop PC is needed in a small form-factor, an Intel NUC mini PC could be used.  

If you would like to explore this possibility, bundle deals for the 400 Series cameras and Up Board or NUC are both available in the RealSense section of Intel's online click store.

https://click.intel.com/realsense.html

Here's a specification sheet for the D4.

https://www.mouser.com/pdfdocs/Intel_Vision_Processor_D4_ProductBrief.pdf

pierref33, 2018-07-19:
This totally answer my question, Thank you!!!

closed 2018-07-19 by pierref33