realsenseai/librealsense #2088: post processing code in python

pierref33, 2018-07-19:
Hello,

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Win (10)   |                                     | 
| Platform                                 | PC  |
| SDK Version                            |    2.12.0                          | 
| Language                            | opencv/python 2.7                         | 

### Issue Description
I don't see any post processing code in python for the camera like in C++. Is there any way to implement an existing one (in a python script), or should I try to create it from scratch?

Thank you in advance !

MartyG-RealSense, 2018-07-19:
Post-processing via the Python API was added in version 2.11.0 of RealSense SDK 2.0.  Although I am not aware of the existence of official documentation for Python post-processing, I did find an example script that a user provided.

https://github.com/IntelRealSense/librealsense/issues/1672

pierref33, 2018-07-19:
Thank you, it helps me a lot, when I test it it seems to work!

However, I am not an expert in computing and so I didn't understand the answer :
 "You should be able to set alpha/delta parameters using the regular options API on filters that support them"

Is it possible for you to explain me how to change these parameters simply ( preferably with an example) in order to change these parameters directly on my python 2.7 code.
Thank you.

MartyG-RealSense, 2018-07-19:
Alpha and delta parameters can provide smoothing of the image.  The post-processing documentation provides more details of using alpha smoothing with the function RS2_OPTION_FILTER_SMOOTH_ALPHA

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md

Delta smoothing has an instruction called RS2_OPTION_FILTER_SMOOTH_DELTA

I could not find any Python-specific code examples of using alpha and delta smoothing though

dorodnic, 2018-07-19:
This would be the perfect opportunity to test our newly added [Jupyter Notebooks](https://github.com/IntelRealSense/librealsense/tree/jupyter) infrastructure.
You can review the example [here](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb) or run it in an interactive cloud container by clicking:
[image]

MartyG-RealSense, 2018-07-19:
I tried it out myself and both formats of the example worked fine for me.  It looks great.  Congrats, and thanks for you and your team's continual excellent work, Dorodnic.

pierref33, 2018-07-20:
I will test it as soon as possible. This is some great examples to see more option for the camera in python. Thank you for adding it and MartyG-RealSense  thank you to have answer to all my questions.

RealSense-Customer-Engineering, 2018-07-31:
[Realsense Customer Engineering Team Comment]
@pierref33
Please download the file as "python", then run with the command python distance_to_object.py.

carypai, 2018-07-31:
[image]
[image]

closed 2018-08-14 by RealSense-Customer-Engineering

dienhoa, 2019-01-24:
Hi @pierref33  . Where did you download the object_detection.bag file ? It is not downloaded automatically in the notebook like others ? Thank you in advance

MartyG-RealSense, 2019-01-24:
I believe it is downloaded from the link below.

http://realsense-hw-public.s3.amazonaws.com/rs-tests/TestData/object_detection.bag

The link to the file originated from here:

https://colab.research.google.com/drive/10YTLAf2i0R80-XX_6-1gTPtOMD0cRIx5#scrollTo=Z0UntaQv3BtP

dienhoa, 2019-01-24:
Ah thank you @MartyG-RealSense . I am thinking doing the same with the amazonaws . Especially thank you again for guiding me here :D I read your post on Intel Forum and it's very helpful that I am more familiar working with Jupyter

MartyG-RealSense, 2019-01-24:
If you go to the link below, it's the 4th link down in the list.  you can either left-click on it to view the contents or right click on it and select Save Link As, or Save Target As, or whatever your particular browser calls the option to download a link.

https://github.com/chuanqi305/MobileNet-SSD/find/master