realsenseai/librealsense #2089: CustomCalibrationTest sln error

BGA3, 2018-07-19:
Hi 

First of all, i'm not sure that this is the right place to post but here it goes.

For the CalibrationToolAPI (both v. 2.5.2.0. and 2.6.4.0) the CustomCalibration visual studio project is not correctly set up. It complains about several .lib files that is NOT present in OpenCV 3.3.0 (which is listed in the documentation as being the openCV version to use). I have used quite some time on fixing this, but with no luck. 

Can you please fix this, and provide a CalibrationToolAPI that actually works?

if relevant, I am working on windows 10 x64, visual studio 2017.

BGA3, 2018-07-19:
Edit: I am building Release in x64. The first issue is the "zlib.lib"-file, which is not part of the OpenCV 3.3.0. With my knowledge, this .lib file is only present in the OpenCV 2.x.x.. As such, I believe you are mixing up the openCV versions.

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
hi @BGA3,

do you mean you met compile error? i just build it as x64-release and seems no problem, so can you please tell your steps so i can try to reproduce?
thanks.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @BGA3,
do you still having this issue?

closed 2018-08-23 by RealSense-Customer-Engineering