realsenseai/librealsense #2090: How do I convert depth image to point cloud?

impet14, 2018-07-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   linux  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

How do I convert depth image to point cloud? Now, I am doing object detection from color then I get the region on the depth image. I would like to convert dept image to PC. Can anyone give me suggestion?
Thanks

summer1216, 2018-07-19:
In **examples** of this library there's point cloud generation in OpenGL. In **wrappers** there's wrapper to convert realsense point could into pcl point cloud.

MartyG-RealSense, 2018-07-19:
Have you seen the step by step tutorial on generating a point cloud?

https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

impet14, 2018-07-20:
Do we have python version of example?

MartyG-RealSense, 2018-07-20:
There is an example for saving point clouds as PLY files directly from Pyrealsense.

https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/export_ply_example.py

impet14, 2018-07-26:
How can we visualize point cloud using python?

MartyG-RealSense, 2018-07-26:
I searched very hard but was unable to find a good, clear example of creating a point cloud in Python.  One of the Intel support people on this forum such as Dorodnic or RealSense-Customer-Engineering may be able to answer this better, as they have access to resources that I do not.

The best I could find was a page in Japanese, where they apparently create a Python point cloud with a D415 by importing the Point Cloud Library (PCL).

https://qiita.com/mizumasa/items/c6ca2c73fa86c9a5a39c

baptiste-mnh, 2018-07-27:
I'm also searching an efficient way to visualize in real time the generated pointcloud from the D400 cam from python wrapper.
It's possible with the open3D visualizer but way too slow for now. I recently add two functions in the development branch of librealsense SDK to make it works. : https://github.com/IntelVCL/Open3D/issues/473#issuecomment-408339685

You can have a look at [python-pcl](https://github.com/strawlab/python-pcl) which is a python binding of  the PCL lib but I did not manage to install it...

So I'm trying to find a way to create my own viewer using OpenGL and the python wrapper by getting the vertices and textures coordinate in numpy.

If anyone have a clue about how to do it or a better way to visualise the point clouds It could be very usefull!

dorodnic, 2018-07-30:
@baptiste-mnh thank you for the feedback, looking forward to your findings.
In parallel, I've been looking at point-cloud visualisation inside Jupyter notebooks using [pyntcloud](https://github.com/daavoo/pyntcloud):
[
[image]
](https://mybinder.org/v2/gh/dorodnic/binder_test/master?filepath=pointcloud.ipynb)

MartyG-RealSense, 2018-07-30:
Thanks muchly for the point cloud tutorial, Dorodnic - I'll make sure the link gets promoted and passed on to those on the RealSense Support forum who have been asking about Python point clouds.

RealSense-Customer-Engineering, 2018-08-07:
[Realsense Customer Engineering Team Comment]
@impet14
Still have any question related to this topic?

closed 2018-08-14 by RealSense-Customer-Engineering