realsenseai/librealsense #2093: How to estimate depth from infrared frame?

summer1216, 2018-07-19:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version | Linux | 
| Kernel Version (Linux Only)    |  4.4.0-127                                       | 
| Platform                                 | PC  |
  

I have a somehow general question, I have a D435, and I can have color, depth, and infrared frames. By using the library, we can align the depth and color image and get the depth value from the depth frame by using functions like get_distance. However, I am confused how can we utilize the infrared imaged based on the structured light? How can we get the 3D distance of a pixel in infrared frame?

It is written the depth range for D435 is from 11cm, but the depth range from stereo depth frame is something like 30cm. So I assume there should be some method to get the distance value from infrared frame. 

My working senario needs to work in a range like 10cm -30 cm. That's the reason why I bought a D435.

Thank you!

MartyG-RealSense, 2018-07-19:
I have not heard of a way to get distance from the infrared frame alone without using depth (though that does not mean it does not exist).

If I were approaching the problem myself, I would alter the depth scaling factor of the D435, as its default settings are not as well suited to close-range scanning as the earlier SR300 RealSense model.  The principle of why this is so, and how the scaling factor of the camera may be adjusted to be closer to that of the SR300, is discussed in the link below, and another link you can visit once you reach that page.

https://communities.intel.com/message/553939#553939

RealSense-Customer-Engineering, 2018-07-20:
[Realsense Customer Engineering Team Comment]
@summer1216
Since our camera use pixel disparity to calculate the depth, so if you want to reach to around 10cm using D435, you need to lower down the resolution to 424*240. With higher resolution, the minimum distance will be longer.

MartyG-RealSense, 2018-07-20:
Yes, the advice of RealSense Customer Engineering concurs with what the link I provided said.

"A cruder method of increasing close-range accuracy is apparently to reduce the resolution to a low one, less than 848x400"

summer1216, 2018-07-20:
Thank you very much for all your replies. @MartyG-RealSense @RealSense-Customer-Engineering There are a few things I am still confused. 

1. Lower down the resolution means both color and depth streams, or only depth? I want to align both streams, and I want to keep the largest color resolution. 

I have an SR300 as well, I purchased one more D435 because of its better depth performance in short range. I hope this will work. 

2. What can we do by using the infrared image? It seems like now we can only visualize it. I don't know what can we get from this special frame, I thought it is also for distance measurement, but it seems not.

MartyG-RealSense, 2018-07-20:
1.  If you plan on aligning the color and depth streams, I recommend setting color and depth resolution to the same whenever possible.

2.  The data sheet document for the 400 Series cameras has the following explanation of how it creates depth.

*******

The Intel RealSense depth camera D400 series uses stereo vision to calculate depth.  The stereo vision implementation consists of a left imager, right imager, and an optional infrared projector.  The infrared projector projects non-visible static IR pattern to improve depth accuracy in scenes with low texture. 

The left and right imagers capture the scene and sends imager data to the depth imaging processor, which calculates depth values for each pixel in the image by correlating points on the left image to the right image and via shift between a point on the Left image and the Right image. 

The depth pixel values are processed to generate a depth frame.  Subsequent depth frames create a depth video stream.

0xLeon, 2018-07-20:
You can't get depth from one IR image alone. The D415 and D435 do not use structured light with the IR projector. The pattern is only used to enhance depth perception for the implemented Semi-Global Matching. This is especially necessary for low to non textured surfaces. I did some tests for a university project regarding the laser projection: You are not able to get any depth from a completely flat, untextured surface without the laser enabled.

RealSense-Customer-Engineering, 2018-07-30:
[Realsense Customer Engineering Team Comment]
@summer1216
Do you still have question? I think developers have answered your question.

summer1216, 2018-08-02:
@RealSense-Customer-Engineering thank you very much! And this issue can be closed.

closed 2018-08-03 by RealSense-Customer-Engineering