realsenseai/librealsense #2094: realsense_viewer with automatic control

TAREK-ELOUARET, 2018-07-19:
Hello Everyone,

As I am a new in field of programming and c++, all what I want is, what's the differents changes to make in order to open realsense_viewer programm cpp with an fixed control and record ( to save a bag) ( without interacting by users).

Thanks in advance.

AntoineWefit, 2018-07-19:
I don't really see the point of opening the RealSense Viewer if you don't want to change the parameters, so I assume you mean "rendering a stream with fixed parameters" instead, am I right ?

I would recommend you check out the examples codes : 
https://github.com/IntelRealSense/librealsense/tree/master/examples

Maybe try combining :
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/imshow/rs-imshow.cpp
with
https://github.com/IntelRealSense/librealsense/blob/master/examples/save-to-disk/rs-save-to-disk.cpp
to render the stream and save a file (again, I'm assuming that's what you want to do).

And if you are beginning, take a look at :
https://github.com/IntelRealSense/librealsense/wiki/API-How-To

dorodnic, 2018-07-19:
Hi @TAREK-ELOUARET 
Thank you for your input. Writing dedicated C++ / python code is probably better long term solution.
Opening the Viewer to limited automation is an interesting idea (that we also explore in need for better testing). We will consider it.

TAREK-ELOUARET, 2018-07-20:
Hi @AntoineWefit 
I made some modifications in record-playback file so I could resolve my problem, instead, I have tried to implement opencv with realsense ( as libraries ) but the result is less reliable (very slow in record and streaming ) than using and exploit realsense parameters. 

So thank's for your help Sir.

@dorodnic  I would like to appreciate your time and consideration, you helped me in lot of issues, so thanks' in advance

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
@TAREK-ELOUARET
If the issue is solved and no further question, please help to close the ticket, thank you.

dorodnic, 2018-07-23:
@RealSense-Customer-Engineering 
I'd like to keep this is open as enhancement request.

closed 2019-02-17 by dorodnic