realsenseai/librealsense #2095: error on reproducing example: no matching function for call to ‘rs2::align::align()’

konnerthg, 2018-07-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |NA | 
| Firmware Version                   | NA | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.13.0-36-generic                                      | 
| Platform                                 | PC |
| SDK Version                            | 2.14.0 }                          | 
| Language                            | C++                          | 
| Segment			|  others                          | 

### Issue Description

I'm trying to reproduce the example from: https://github.com/IntelRealSense/librealsense/blob/master/examples/align/rs-align.cpp

This is the relevant cutout of my code:
```
rs2::pipeline pipe;
rs2::pipeline_profile profile = pipe.start();
float depth_scale = get_depth_scale(profile.get_device());
rs2_stream align_to = find_stream_to_align(profile.get_streams());
rs2::align align(align_to);
```
On build I receive:

```
error: no matching function for call to ‘rs2::align::align()’
 RSCam::RSCam()
              ^
In file included from /usr/local/include/librealsense2/hpp/rs_context.hpp:9:0,
                 from /usr/local/include/librealsense2/rs.hpp:9,
                 from /home/demo/gr/KinectFusion_cuda/KinectFusionClient/include/depth_camera.h:11,
                 from /home/demo/gr/KinectFusion_cuda/KinectFusionClient/src/depth_camera.cpp:2:
/usr/local/include/librealsense2/hpp/rs_processing.hpp:448:9: note: candidate: rs2::align::align(rs2_stream)
         align(rs2_stream align_to) :_queue(1)
         ^
/usr/local/include/librealsense2/hpp/rs_processing.hpp:448:9: note:   candidate expects 1 argument, 0 provided
/usr/local/include/librealsense2/hpp/rs_processing.hpp:436:11: note: candidate: rs2::align::align(const rs2::align&)
     class align
           ^
/usr/local/include/librealsense2/hpp/rs_processing.hpp:436:11: note:   candidate expects 1 argument, 0 provided
/usr/local/include/librealsense2/hpp/rs_processing.hpp:436:11: note: candidate: rs2::align::align(rs2::align&&)
/usr/local/include/librealsense2/hpp/rs_processing.hpp:436:11: note:   candidate expects 1 argument, 0 provided
```
This error doesn't make any sense to me. Where am I wrong?

RealSense-Customer-Engineering, 2018-07-20:
[Realsense Customer Engineering Team Comment]
@Konnerthg,
Can you provide the full code? since from the snippet it looked the same as the sample, and the error code is hard to know exactly where the problem is. did you also encounter the same issue when build our rs-align example?

konnerthg, 2018-07-20:
So i had a header file with the following definition:

```
class RSCam {
private:
    rs2::align _align;
}
```

This was calling the empty constructor which in my C++ inexperience I didn't know. I'm sorry. Thank you for your effort.

closed 2018-07-20 by konnerthg