realsenseai/librealsense #2096: How to set HW sync with multicamera?

rafaor29, 2018-07-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |  Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  4.13.0-45-generic                                         | 
| Platform                                 | PC |

### Issue Description
Hello, i already have all the material necessary to stablish the HW sync in the cameras but i found some trouble since there is no example of how to do it.l
I would appreciate if someone could enlighten this a little bit, thank you.

MartyG-RealSense, 2018-07-19:
Intel have published a White Paper that describes the sync process.  Have you seen it?

https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper_rev1.1.pdf

rafaor29, 2018-07-20:
Yes i've seen it but i cannot set the parameters. I have the following code and i'm finding problems to convert the rs2::device dev to rs2_device* to send it to the module for HW sync. 

 ` void enable_device(rs2::device dev)
    {
        std::string serial_number(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        std::lock_guard<std::mutex> lock(_mutex);

        if (_devices.find(serial_number) != _devices.end())
        {
            return; //already in
        }

        // Ignoring platform cameras (webcams, etc..)
        if (platform_camera_name == dev.get_info(RS2_CAMERA_INFO_NAME))
        {
            return;
        }
        // Create a pipeline from the given device
        rs2::pipeline p;
        rs2::config c;
        c.enable_device(serial_number);
        // Start the pipeline with the configuration
        rs2::pipeline_profile profile = p.start(c);
        // Hold it internally
        _devices.emplace(serial_number, view_port{ {},{},{}, p, profile });

        // set camera as master
        rs2_error* e = nullptr;
        const std::shared_ptr<rs2_device> device = dev.get();
        uint8_t input[24] = {0};
        input[0] = 0x14;
        input[2] = 0xab;
        input[3] = 0xcd;
        input[4] = 0x64;
        input[8] = 0x1;
        auto result = rs2_send_and_receive_raw_data(device, static_cast<void*>(input), 24, &e);
        rs2::error::handle(e);

    }`

MartyG-RealSense, 2018-07-20:
I see you have the latest version of the firmware.  Do you also have a recent version of RealSense SDK 2.0 please?  The hardware sync support in the firmware is only half of what is needed.   The other half of the sync system is in the SDK, in versions 2.13.0 and newer.

rafaor29, 2018-07-20:
The version of the SDK is 2.14.0

MartyG-RealSense, 2018-07-20:
What catches my notice in the script above is that on the first line, you have a ' in front of void.  Is that a typo in this forum message or do you have that in your actual script?

You may also find the discussion link below a useful read.

https://communities.intel.com/message/531970#531970

hinmanj, 2018-07-21:
I got the hardware sync via one master and one slave D415 using the set_option with: RS2_OPTION_INTER_CAM_SYNC_MODE

I was originally trying it your way @rafaor29 but it wasn't working, so when updating to 2.14.0 I looked through a bit and saw things seem to work through the set_option now instead. I think syncing color is still broken, however, but haven't tested.

I'm actually trying to use an external trigger for the hardware sync now, with the cameras set as slaves, and it's seemingly partially working, but not usably-so. I have an external pulse at 1.8v at 30hz, with a positive pulse of 2 milliseconds (much more than the required 100 microseconds specified in the paper), and when I run my slave at 30fps, it very sporadically updates the depth frames (definitely not at 30hz, nor even at a regular interval).

Anyone tested the external sync trigger with the latest firmware and 2.14.0 yet? I hope to grab an oscilloscope next week and see what the signal coming out of the master-D415 that triggers the slave when it *does* work, and then compare it with my external pulse signal.

hinmanj, 2018-07-21:
Relating to my previous comment, I notice in the paper this paragraph:

> However, it is important to make sure to use a high resolution signal generator. The frequency of the
signal generator needs to exactly match the sensor frame rate. For example, if the sensor is set up as
30 fps, the real frame rate may be 30.015 fps. You may need to use an oscilloscope to measure the
real frame and configure the signal generator to the same frequency. 

My external pulse is not **perfectly** 30fps, and does very **ever so slightly** between pulses. Is that truly and issue, and if so, why? Is the _master D415_ in the master<->slave situation, when set to 30fps, truly **exactly 30fps as a signal generator**?

agrunnet, 2018-07-21:
Correct. The image sensors we use from Omnivision do not permit very much deviation from their expected frame rates. This is why we state that a good external sync is actually to use the master from another camera even if you don’t use that camera for anything.

hinmanj, 2018-07-21:
At this point I'm probably hijacking the thread, but after fixing my imperfect 30hz external trigger pulse, the D415 seems to work now, but with the caveat that the depth is behaving strangely, [https://youtu.be/mLtbCX2RG28](here is an example video.)

Essentially the depth quality is dying, which I can only assume is due to the trigger signal as I didn't notice this when slaving to another D415. It doesn't seem to be the projector having issues either due because when I cover it, it has no affect on the depth in my current scenario.

rafaor29, 2018-07-23:
Hi @hinmanj , the way i did it i wasn't able to find a pulse when the camera was supposed to be generating it. 

I was trying to do it with the RS2_OPTION_INTER_CAM_SYNC_MODE that you talked about but i didn't understand how you do it exactly. I was trying the program attached. 

[rs-multicam.zip](https://github.com/IntelRealSense/librealsense/files/2219818/rs-multicam.zip)

``

hinmanj, 2018-07-23:
I only looked over this for a couple seconds, so forgive me if I'm missing something, but this snippet of yours:

`    int cont = 0;

    for (auto&& dev : ctx.query_devices()) // Query the list of connected RealSense devices
    {
        if( cont == 0){
            connected_devices.enable_device(dev, 1);
        }
        else{
            connected_devices.enable_device(dev, 2);
        }
    }
`

Wouldn't that result in all connected devices being set as master?

closed 2018-07-30 by rafaor29