realsenseai/librealsense #2102: Why does the frame drop occur when record color frame and depth frame.

suho-j, 2018-07-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|-------------------------------- |------------------------------------------- |
| Camera Model                       | { D435 }                                                  | 
| Firmware Version                   | (newest)                                                 | 
| Operating System & Version |   {Win 10                                                | 
| Platform                                 | PC/ Zotac barebone PC                         |
| SDK Version                           |  { librealsense 2.10.4 }                            | 
| Language                               |  {opencv/python }                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Why does the frame drop occur when record color frame and depth frame.
The program I want is as follows.
First, record color frame and depth frame at 60fps
second, color and depth frames are called for image processing.
However, filter application and recording will cause the frame drop
How can i fix this problem?
please help.. 

source code
--------------------------------------
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 60)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 60)
# config.enable_device_from_file("workd.bag")
_recoder = config.enable_record_to_file("./work.bag")
pipeline.start(config)

# # dec_filter = rs.decimation_filter()
 # spat_filter = rs.spatial_filter()
 # temp_filter = rs.temporal_filter()
 # disparity = rs.disparity_transform()

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

#        # 30fps
#        # filtered = dec_filter.process(depth_frame)
#        # 13fps
#        # filtered = spat_filter.process(depth_frame)
#        # 53fps
#        # filtered = disparity.process(depth_frame)
#        # 60fps
#        # filtered = temp_filter.process(depth_frame)

         # filtered_image = np.asanyarray(filtered.get_data())
         depth_image = np.asanyarray(depth_frame.get_data())
         color_image = np.asanyarray(color_frame.get_data())

         depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
#         # filtered_colormap = cv2.applyColorMap(cv2.convertScaleAbs(filtered_image, alpha=0.03), cv2.COLORMAP_JET)
         colormap = cv2.applyColorMap(cv2.convertScaleAbs(color_image, alpha=0.03), cv2.COLORMAP_JET)

         # Stack both images horizontally
         images = np.hstack((depth_colormap, colormap))

         cv2.imshow('video', images)

 finally:
     # Stop streaming
     pipeline.stop()

MartyG-RealSense, 2018-07-20:
The subject of slow running speed or missed frames has come up before on the other RealSense support forum.  The discussion linked to below may be useful to you.

https://communities.intel.com/message/553003#553003

suho-j, 2018-07-25:
Thank you for your reply.
I went to the link and i read it.
but There is only a C++ sample in the content.
Could you tell me the sample implemented as a python??

MartyG-RealSense, 2018-07-25:
I am not personally experienced in Python programming unfortunately.  There is a general guide to For loops in Python though that may provide some guidance about how to convert the code for Python.

https://wiki.python.org/moin/ForLoop

RealSense-Customer-Engineering, 2018-08-14:
[Realsense Customer Engineering Team Comment]
hi @suho-j,
would like to know if MartyG's suggestion work for you? thanks.

RealSense-Customer-Engineering, 2018-10-08:
[Realsense Customer Engineering Team Comment]
Hi @suho-j,

Still observe this issue? If not, will close this.

dorodnic, 2018-11-08:
The playback issue should be addressed.
We also noticed that on some low-end systems, the recorder can't keep up despite working with an SSD because the CPU doesn't handle the built-in frame compression. Usually compression is preferred because the bottleneck is the hard-drive, but we are adding an API to control it (#2687).
In any case, if you have follow-up questions or encounter any problem feel free to open new issues.

closed 2018-11-08 by dorodnic