realsenseai/librealsense #2103: Using colorize_frame in modified threaded multi-cam example

ahpuck, 2018-07-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | (5.9.14) | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.13.0                      | 
| Language                            |  C++                          | 
| Segment			|  Others                          | 

### Issue Description
Whenever I call colorize_frame within a thread for the multicam example (see psuedo-code below) within seconds of the application starting the program crashes. I can see a couple of frames get through in the application window and then a crash happens generally somewhere in rs_processing. Happy to provide more info if someone can point me in the right direction! 

At this point I've tried moving the colorizer to a downstream messaging queue (which is also threaded) and it also crashes. Basically anytime I call colorizer from a thread, the program crashes. If I keep in downstream (like right before rendering) everything seems to work fine. I really only need the colorized video frames from the depth data and after that I don't need depth data anymore. 

Here's the basis of my code (forked from multi-cam example)

device_container.h
```
// helper struct for each camera's pipeline
struct view_port {
	std::map<int, rs2::frame> frames_per_stream;
	texture tex;
	rs2::pipeline pipe;
	rs2::pipeline_profile profile;

	std::string serial_no;
	// create the frame queues
	rs2::frame_queue processed_frames;
	// create the message queues
	rs2::frame_queue message_frames;
	std::vector<frame_filter_options> filters;
	rs2::colorizer colorize_frame;
};

class device_container
{
public:
        // code for this is identical to multi-cam example except for the below functions
	std::vector<std::thread*> threads;
	void _poll_processed_queue(view_port *view, int stream_no);
```

device_container.cpp
```
void device_container::_poll_processed_queue(view_port *view, int stream_no) {
	// Ask each pipeline if there are new frames available
	rs2::processing_block frame_processor(
		[&](rs2::frameset data,        // Input frameset (from the pipeline)
			rs2::frame_source &source) // Frame pool that can allocate new frames
	{

		rs2::depth_frame filtered_frame = data.get_depth_frame();

		for (auto&& filter : view->filters) {
			if (filter.is_enabled) {
				filtered_frame = filter.filter.process(filtered_frame);
			}
		}

		auto vid_frame = view->colorize_frame(filtered_frame);
		view->message_frames.enqueue(vid_frame);
		rs2::frameset combined = source.allocate_composite_frame({ vid_frame });
		source.frame_ready(combined);

	});
	frame_processor >> view->processed_frames;
	while (_config->alive) {
		rs2::frameset frameset;
		if (view->pipe.poll_for_frames(&frameset)) {
			frame_processor.invoke(frameset);
		}
	}
}
```

main.cpp
```
int main() {
// device_container initialization as seen in multi-cam example.
	int stream_no = 0;
	for (auto&& device : connected_devices._devices) {
		auto *view = &device.second;
		std::thread *video_thread = new std::thread(&device_container::_poll_processed_queue, &connected_devices, std::ref(view), stream_no);
		connected_devices.threads.emplace_back(video_thread);
		video_thread->detach();
		stream_no++;
	}
// rest of code basically multicam example to render window.
}
```

ahpuck, 2018-07-20:
Okay so I've modified the color map code to check for NAN and it's stopped crashing.
```
float3 color_map::get(float value) const
	{
		if (_max == _min)
		{
			return *_data;
		}
		float t = (value - _min) / (_max - _min);
		t = clamp_val(t, 0.f, 1.f);
                 if(isnan(t)){
                   return *_data
                 }
		return _data[(int)(t * (_size - 1))];
	}
```

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
@ahpuck
Since the RealSense viewer supports multiple cameras with colorized feature without problem, can you share the full code for me to reduplicate the issue? Guessing might be something wrong after you apply the filter then do the colorization, with the full code package will be easier to find why cause the problem.

ahpuck, 2018-07-26:
Definitely. I'll strip out it down to the bare bones and upload it. The full application is actually quite large but the basic example above has the issue.

ahpuck, 2018-07-27:
here's a gist with the code broken down to the just threading the camera depth processing. 
https://gist.github.com/ahpuck/87bcb0a937a797cb71aacd4b28cbc644

Basically, if I move colorize_frame out of _poll_processed_frames (the threaded video processing) and into poll_frames the code works just fine. I

ahpuck, 2018-07-27:
Sorry I mean _poll_processed_queue not _poll_processed_frames

ahpuck, 2018-08-03:
Would the complete visual studio solution be of any help?

RealSense-Customer-Engineering, 2018-08-06:
[Realsense Customer Engineering Team Comment]
@ahpuck
Sure, if you can attach the solution file which can save our time.

RealSense-Customer-Engineering, 2018-08-24:
[Realsense Customer Engineering Team Comment]
will you upload the solution file?

dorodnic, 2018-11-27:
@ahpuck - do you still need help with this issue?

ahpuck, 2018-11-27:
No please close. Thank you.
Austin Puckett
(970) 231-4364
austinhpuckett@gmail.com

closed 2018-11-28 by dorodnic