realsenseai/librealsense #2107: csharp for osx

bleppie, 2018-07-20:
The csharp wrapper can't be built on OSX because it seems to rely on Visual Studio and won't use Mono.

If you have access to Windows, there is a workaround found here (https://communities.intel.com/thread/122475), but folks with only OSX can't build the necessary wrappers/csharp/Intel.RealSense/Intel.RealSense.csproj for the workaround.

Please provide a way to build on Mono, or include the prebuilt dll and bundle  in the repo.

Thanks!

MartyG-RealSense, 2018-07-20:
Although not ideal for those who want to use Mono, there is a free Community Edition of Visual Studio for OSX.

https://visualstudio.microsoft.com/vs/mac/

For Windows users who have been having trouble with compiling the DLLs for use in Unity, the method of dragging-and dropping the pre-built DLLs in the bin folder instead of compiling from source does seem to be working consistently.  I have no information about whether the drag-and-drop technique is effective on OSX though.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @bleppie,

do you need more support on this topic?

bleppie, 2018-08-10:
I tried Visual Studio for Mac but unfortunately the cmake files have
hard-coded checks to prevent csharp builds on Mac. I tried removing them
and wrestled with it a bit but couldn't get it to work. I've now moved
Development over to Windows so this isn't a pressing issue for me, but I
would still like to see support for Mac users.

Thanks

RealSense-Customer-Engineering, 2018-08-15:
[Realsense Customer Engineering Team Comment]
hi @bleppie,

yes, i have a try too and can't make it work on Mac either.
i think this is good to have, and we will improve the wrappers time to time.
let me close this ticket for now and please feel free to open new ticket if any question.

closed 2018-08-15 by RealSense-Customer-Engineering