realsenseai/librealsense #2109: Changing image size in RealsenseStreamTexture.ProcessFrame

bleppie, 2018-07-20:
I'm subclassing RealsenseStreamTexture and overriding ProcessFrame to pass it through a bunch of filters. If one of those filters changes the image size (a decimate filter, for example), I'd like to resize the texture as well, but there doesn't seem to be an elegant way to do this.

I am able to access the texture member because it is marked protected, but I can't do much with it in ProcessFrame because ProcessFrame might not be called in the main unity thread. Calling texture.width, for example, results in the error "GetDataWidth can only be called from the main thread".

I could try changing the texture in Update, but  RealsenseStreamTexture.Update is private so I can't call it, and the code in there uses private members, so I can't recreate it.

Instead, in ProcessFrame I store the frame width and height and then use LateUpdate to check if the texture needs to be recreated. This seems to work but is hackish. (I am having problems with Unity not recognizing the change until I deactivate and reactivate any components that use the changed shader, but this seems to be a Unity problem).

It would be nice if RealsenseStreamTexture checked for size changes after calling ProcessFrame and resized the texture appropriately.

Cheers

bleppie, 2018-07-21:
A little more info for the part I elided over: "I am having problems with Unity not recognizing the change until I deactivate and reactivate any components that use the changed shader"

If anyone comes across this, it's because directly changing the texture in the material (binding the TextureEvent  to depthMat.Material.maintexture, as in the examples) changes the shared material, not the copy of it made for the raw image renderer. To get around this, change the texture in the RawImage (bind the TextureEvent to DepthImage.RawImage.texture).

It might be helpful to change the examples to do this, or put it in the docs somewhere. It will also come in handy if someone has multiple cameras and wants to use the same material for all of them.

matkatz1, 2018-07-24:
Thanks for your comment.
I'm currently working on a slightly different approach for frame processing in which you can provide processing blocks to the device that performs as processing pipe.

The changes can be found here:
[https://github.com/matkatz1/librealsense/tree/unity_processing_blocks](https://github.com/matkatz1/librealsense/tree/unity_processing_blocks)

A sample scene named "PointCloudProcessingBlocks", demonstrates the use of the filters, it is currently doesn't include the decimation filter since  it require modification to the pointcloud generator but you can add that filter to any of the depth texture scenes.
[image]

bleppie, 2018-07-26:
Ah great, that approach looks better. One thing I would suggest is to set the filter options in Start, OnValidate, and when they are changed programmatically rather than on every frame in Update.

matkatz1, 2018-07-31:
I'm not sure that OnValidate is right for this case since it will respond only for changes that are done in the inspector: "Description: This function is called when the script is loaded or a value is changed in the inspector (Called in the editor only)."

bleppie, 2018-08-02:
Yup, generally you need to set parameters in Start, for the initial values,
OnValidate, in case the user changes them from the inspector, and from any
programmatic property changes. A bit more work, but much of this you can do
in the base class.

RealSense-Customer-Engineering, 2018-08-06:
[Realsense Customer Engineering Team Comment]
@bleppie
Does the way provided by matkatz1 solves your problem?

@matkatz1
Please consider to raise PR for the implementation if you want. Thank you.

bleppie, 2018-08-10:
It should as long as it handles decimation

RealSense-Customer-Engineering, 2018-08-24:
[Realsense Customer Engineering Team Comment]
@bleppie
Don't understand what do you mean for "it should as long as it handles decimation"? is there still function needs to be implemented?

bleppie, 2018-08-24:
I just checked out the latest version and it works, so you can close this
issue. yay! (The previous version didn't update textures correctly when a
decimation filter was inserted into the pipeline).

RealSense-Customer-Engineering, 2018-08-24:
[Realsense Customer Engineering Team Comment]
Glad to hear that. Enjoy the coding!!

closed 2018-08-24 by RealSense-Customer-Engineering