realsenseai/librealsense #2110: Select ROI in RealSense Viewer

apoorva2398, 2018-07-20:
Please support selecting a region of interest (ROI) in RealSense Viewer (in addition to the mouse wheel for zooming in and the left mouse button to pan), like the ROI for auto exposure.

dorodnic, 2018-07-20:
Hi @apoorva2398 
Could you elaborate what you mean by ROI? What would be the functionality of this rect?

apoorva2398, 2018-07-20:
e.g. if I wanted to select the forehead from a still person's face and
spatially average it---basically, cropping

Some cameras like the Rolera EM-C2 get an increase in fps with smaller ROI.

RealSense-Customer-Engineering, 2018-07-24:
[Realsense Customer Engineering Team Comment]
hi @apoorva2398,
so far we support fixed FPS as listed on datasheet only, so i think we can't increase FPS with smaller ROI.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @apoorva2398,

do  you need more support on this?

closed 2018-08-27 by RealSense-Customer-Engineering