realsenseai/librealsense #2111: Run after build on Mac OSX

connorgoggins, 2018-07-21:
I read the documentation  and finished building the SDK for Mac OS, but am having trouble figuring out how to run the build as a desktop application. I located the build output in the DerivedData directory, but within that directory are two more directories (librealsense2/ and ModuleCache.noindex/). I've looked everywhere online and can't find any solution for converting these directories into a .dmg or .app file. How should I go about converting this build output into a format that is runnable on a Mac?

MartyG-RealSense, 2018-07-21:
The SDK itself does not run as a launchable application, even on Windows.  Instead, programs created with the SDK (such as sample programs, and SDK tools like the RealSense Viewer) are the ones that are run.

Whilst I have no experience first-hand in building Librealsense apps on OSX with XCode, if I were doing it, in theory, then I would try using the standard methods of exporting a project from XCode.

The link below seems like a good description of how to export a dmg to your computer and not to the App Store.

https://apple.stackexchange.com/questions/174008/xcode-how-to-export-app-solution

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @connorgoggins,

do you need more support on this topic?

closed 2018-08-21 by RealSense-Customer-Engineering