realsenseai/librealsense #2112: Rectified images Y alignment

timprepscius, 2018-07-21:
D435

--

I don't know whether this is a bug or not a bug.  It doesn't affect me, I think, however it is a bit strange.

I'm seeing that the infrared image 0 and infrared image 1, rectified (Y8) are off in Y alignment slightly.
And always in exactly the same amount (at least for pixels near the center of the image, I haven't tested farther out).

Is this to be expected?  My understanding of the D4 (at least from your documents) is that it is assumes perfect Y alignment, allowing it discard the Y axis while searching for matching pixel intensity changes on the X axis.

Note:  I am not using the RS library for transforming from one image to another, I'm just looking at pixels of the two images.

Anyhow.  Not really an issue.  Just something I'm seeing.

dorodnic, 2018-07-26:
Hi @timprepscius 
Stereoscopic solutions tend to lose calibration overtime. If you are using an engineering sample or if the peripheral was dropped, it might lose calibration a bit, that can explain this behaviour. 
We offer [dynamic calibration software](https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html) to help deal with this problem.

RealSense-Customer-Engineering, 2018-08-02:
[Realsense Customer Engineering Team Comment]
Hi @timprepscius,

Did you try dynamic calibration and see how it works for this issue?

timprepscius, 2018-08-07:
I did not.  I believe I am compensating in software for the calibration problems.  Maybe I'll eventually need to.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
Hi @timprepscius,

I will close this first. If any further issue, you still can re-open or create another ticket.

closed 2018-08-09 by RealSense-Customer-Engineering