realsenseai/librealsense #2116: .getdistance() after post-processing?

v1et4nh, 2018-07-22:
Hi,

how can I use the getdistance()-function after spatial-filtering, temporal filtering, etc. in Python?
The input is a depth-frame, but the output after filtering is only a frame, not a depth-frame anymore. So it doesn't have the getdistance()-function. Is there a way to get the depth information after filtering?

Thank you in advance!

lramati, 2018-07-22:
You can use the `as_depth_frame()` function for this purpose. Keep in mind
that this will only work on a frame that was originally a depth frame. It
can't magically generate depth data for a color frame

dorodnic, 2018-07-26:
@v1et4nh did it work with as_depth_frame?

v1et4nh, 2018-08-05:
Yes, after the running through all the filters, you can use it like this:
depth = frames.get_depth_frame()
filtered = depth_to_disparity.process(depth)
filtered = filtered.as_depth_frame()
distance = filtered.get_distance(x_pos, y_pos)

RealSense-Customer-Engineering, 2018-08-06:
[Realsense Customer Engineering Team Comment]
@v1et4nh
seems your problem is solved, can I close the ticket?

closed 2018-08-06 by v1et4nh