realsenseai/librealsense #2117: Package 'wayland-egl', required by 'gdk-3.0', not found

KeCh96, 2018-07-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   |  | 
| Operating System & Version |   Debian  | 
| Kernel Version (Linux Only)    |  4.11.0 | 
| Platform                                 | ARM board |
| SDK Version                            |  SDK2  | 

I want to build realsense camera on RK3399 ARM board. When I type `cmake ../ -DBUILD_PYTHON_BINDINGS=true`, I run into an error:
```
-- Info: REALSENSE_VERSION_STRING=2.13.0
-- Checking for module 'gtk+-3.0'
--   Package 'wayland-egl', required by 'gdk-3.0', not found
CMake Error at /usr/local/share/cmake-3.12/Modules/FindPkgConfig.cmake:436 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/local/share/cmake-3.12/Modules/FindPkgConfig.cmake:602 (_pkg_check_modules_internal)
  tools/realsense-viewer/CMakeLists.txt:25 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/root/librealsense/build/CMakeFiles/CMakeOutput.log".  
```

I have tried `apt-get install libgtk-3-dev`, but the error still occurs. Could you please help me? @dorodnic @zivsha

dorodnic, 2018-07-23:
Hi @KeCh96 
Please try with `-DBUILD_GRAPHICAL_EXAMPLES=false`. The error seem to come from building the Viewer. If all you need is python, you can safely disable this part.

KeCh96, 2018-07-23:
Thank you very much!! It solve the error, and now I have build `pyrealsense.so` successfully. @dorodnic 

However, when I test it, I get `RuntimeError: No device connected`

I use `journalctl -f _KERNEL_SUBSYSTEM=usb` to debug the usb, and when I plug in the camera, I get the following. (Note that I have not plug out yet)
```
Jul 23 09:00:04 firefly kernel: usb 1-1.1: new high-speed USB device number 58 using ehci-platform
Jul 23 09:00:04 firefly kernel: usb 1-1.1: New USB device found, idVendor=05e3, idProduct=0610
Jul 23 09:00:04 firefly kernel: usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jul 23 09:00:04 firefly kernel: usb 1-1.1: Product: USB2.0 Hub
Jul 23 09:00:04 firefly kernel: usb 1-1.1: Manufacturer: GenesysLogic
Jul 23 09:00:04 firefly kernel: hub 1-1.1:1.0: USB hub found
Jul 23 09:00:04 firefly kernel: hub 1-1.1:1.0: 2 ports detected
Jul 23 09:00:05 firefly kernel: usb 1-1.1.2: new full-speed USB device number 59 using ehci-platform
Jul 23 09:00:05 firefly kernel: usb 1-1.1: clear tt 2 (83b0) error -71
Jul 23 09:00:05 firefly kernel: usb 1-1.1: clear tt 2 (83b0) error -71
Jul 23 09:00:05 firefly kernel: usb 1-1.1.2: unable to read config index 0 descriptor/all
Jul 23 09:00:05 firefly kernel: usb 1-1.1.2: can't read configurations, error -71
Jul 23 09:00:05 firefly kernel: usb 1-1.1: clear tt 2 (83b0) error -71
Jul 23 09:00:05 firefly kernel: hub 1-1.1:1.0: hub_port_status failed (err = -71)
Jul 23 09:00:05 firefly kernel: hub 1-1.1:1.0: hub_port_status failed (err = -71)
Jul 23 09:00:05 firefly kernel: usb 1-1.1: Failed to suspend device, error -71
Jul 23 09:00:05 firefly kernel: usb 1-1.1: USB disconnect, device number 58
```

lramati, 2018-07-23:
@KeCh96 which version of Ubuntu and which desktop environment are you using?

KeCh96, 2018-07-24:
I use `-DFORCE_LIBUVC=true`, but `RuntimeError: No device connected` still occurs. @dorodnic

KeCh96, 2018-07-24:
My Linux kernel is 4.4.77. Do you mean I need to upgrade my Linux kernel? @lramati

dorodnic, 2018-07-24:
With `-DFORCE_LIBUVC=true` you shouldn't have to upgrade the kernel. 
We need to get to the point where the camera is visible via `lsusb` or alike. 
The board has USB3, right?

KeCh96, 2018-07-24:
The blue USB should be USB3. I typed `lsusb` and get:

> Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 001 Device 115: ID 04ca:0061 Lite-On Technology Corp. 
> Bus 001 Device 116: ID 413c:2107 Dell Computer Corp. 
> Bus 001 Device 002: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

KeCh96, 2018-07-25:
I restart the computer and it become OK!! @dorodnic 
But after I plug in and plug out the camera several times, I run into `Segment Fault` error. I am confused by this error. I find the only solution is to restart the computer again...

dorodnic, 2018-07-25:
Glad to hear there is progress :) 
If possible, I'd recommend to upgrade to newer (4.13+) kernel and switch back to V4L2 backend (remove `-DFORCE_LIBUVC=true`). V4L2 backend is not known to set-fault easily. 
That said, the origin of the problem is most likely somewhere at the platform USB level.

KeCh96, 2018-07-26:
Thank you, I will follow your advise soon :)

closed 2018-08-07 by RealSense-Customer-Engineering