realsenseai/librealsense #2118: Cannot install pyrealsense2 in Pycharm on Mac Os

yiyunluo07, 2018-07-23:
Hello, I need to install pyrealsense2 in Pycharm for further programming. But it shows:
Could not find a version that satisfies the requirement pyrealsense2 (from versions: )
No matching distribution found for pyrealsense2

MartyG-RealSense, 2018-07-23:
There was a case involving trying to use Pyrealsense2 with Pycharm a few weeks ago.  Intel's RealSense-Customer-Engineering offered some advice in that post that may be useful to you.

https://github.com/IntelRealSense/librealsense/issues/1994

lramati, 2018-07-23:
Could you please provide the output of running the following snippet from
inside your environment?

```python
from pip._internal import pep425tags
for x in pep425tags.get_supported():
  print(x)
```

yiyunluo07, 2018-07-23:
@lramati 
('cp36', 'cp36m', 'macosx_10_13_x86_64')
('cp36', 'cp36m', 'macosx_10_13_intel')
('cp36', 'cp36m', 'macosx_10_13_fat64')
('cp36', 'cp36m', 'macosx_10_13_fat32')
('cp36', 'cp36m', 'macosx_10_13_universal')
('cp36', 'cp36m', 'macosx_10_12_x86_64')
('cp36', 'cp36m', 'macosx_10_12_intel')
('cp36', 'cp36m', 'macosx_10_12_fat64')
('cp36', 'cp36m', 'macosx_10_12_fat32')
('cp36', 'cp36m', 'macosx_10_12_universal')
('cp36', 'cp36m', 'macosx_10_11_x86_64')
('cp36', 'cp36m', 'macosx_10_11_intel')
('cp36', 'cp36m', 'macosx_10_11_fat64')
('cp36', 'cp36m', 'macosx_10_11_fat32')
('cp36', 'cp36m', 'macosx_10_11_universal')
('cp36', 'cp36m', 'macosx_10_10_x86_64')
('cp36', 'cp36m', 'macosx_10_10_intel')
('cp36', 'cp36m', 'macosx_10_10_fat64')
('cp36', 'cp36m', 'macosx_10_10_fat32')
('cp36', 'cp36m', 'macosx_10_10_universal')
('cp36', 'cp36m', 'macosx_10_9_x86_64')
('cp36', 'cp36m', 'macosx_10_9_intel')
('cp36', 'cp36m', 'macosx_10_9_fat64')
('cp36', 'cp36m', 'macosx_10_9_fat32')
('cp36', 'cp36m', 'macosx_10_9_universal')
('cp36', 'cp36m', 'macosx_10_8_x86_64')
('cp36', 'cp36m', 'macosx_10_8_intel')
('cp36', 'cp36m', 'macosx_10_8_fat64')
('cp36', 'cp36m', 'macosx_10_8_fat32')
('cp36', 'cp36m', 'macosx_10_8_universal')
('cp36', 'cp36m', 'macosx_10_7_x86_64')
('cp36', 'cp36m', 'macosx_10_7_intel')
('cp36', 'cp36m', 'macosx_10_7_fat64')
('cp36', 'cp36m', 'macosx_10_7_fat32')
('cp36', 'cp36m', 'macosx_10_7_universal')
('cp36', 'cp36m', 'macosx_10_6_x86_64')
('cp36', 'cp36m', 'macosx_10_6_intel')
('cp36', 'cp36m', 'macosx_10_6_fat64')
('cp36', 'cp36m', 'macosx_10_6_fat32')
('cp36', 'cp36m', 'macosx_10_6_universal')
('cp36', 'cp36m', 'macosx_10_5_x86_64')
('cp36', 'cp36m', 'macosx_10_5_intel')
('cp36', 'cp36m', 'macosx_10_5_fat64')
('cp36', 'cp36m', 'macosx_10_5_fat32')
('cp36', 'cp36m', 'macosx_10_5_universal')
('cp36', 'cp36m', 'macosx_10_4_intel')
('cp36', 'cp36m', 'macosx_10_4_fat32')
('cp36', 'cp36m', 'macosx_10_4_universal')
('cp36', 'cp36m', 'macosx_10_3_fat32')
('cp36', 'cp36m', 'macosx_10_3_universal')
('cp36', 'cp36m', 'macosx_10_2_fat32')
('cp36', 'cp36m', 'macosx_10_2_universal')
('cp36', 'cp36m', 'macosx_10_1_fat32')
('cp36', 'cp36m', 'macosx_10_1_universal')
('cp36', 'cp36m', 'macosx_10_0_fat32')
('cp36', 'cp36m', 'macosx_10_0_universal')
('cp36', 'abi3', 'macosx_10_13_x86_64')
('cp36', 'abi3', 'macosx_10_13_intel')
('cp36', 'abi3', 'macosx_10_13_fat64')
('cp36', 'abi3', 'macosx_10_13_fat32')
('cp36', 'abi3', 'macosx_10_13_universal')
('cp36', 'abi3', 'macosx_10_12_x86_64')
('cp36', 'abi3', 'macosx_10_12_intel')
('cp36', 'abi3', 'macosx_10_12_fat64')
('cp36', 'abi3', 'macosx_10_12_fat32')
('cp36', 'abi3', 'macosx_10_12_universal')
('cp36', 'abi3', 'macosx_10_11_x86_64')
('cp36', 'abi3', 'macosx_10_11_intel')
('cp36', 'abi3', 'macosx_10_11_fat64')
[... 364 lines cut ...]
('py3', 'none', 'macosx_10_5_intel')
('py3', 'none', 'macosx_10_5_fat64')
('py3', 'none', 'macosx_10_5_fat32')
('py3', 'none', 'macosx_10_5_universal')
('py3', 'none', 'macosx_10_4_intel')
('py3', 'none', 'macosx_10_4_fat32')
('py3', 'none', 'macosx_10_4_universal')
('py3', 'none', 'macosx_10_3_fat32')
('py3', 'none', 'macosx_10_3_universal')
('py3', 'none', 'macosx_10_2_fat32')
('py3', 'none', 'macosx_10_2_universal')
('py3', 'none', 'macosx_10_1_fat32')
('py3', 'none', 'macosx_10_1_universal')
('py3', 'none', 'macosx_10_0_fat32')
('py3', 'none', 'macosx_10_0_universal')
('cp36', 'none', 'any')
('cp3', 'none', 'any')
('py36', 'none', 'any')
('py3', 'none', 'any')
('py35', 'none', 'any')
('py34', 'none', 'any')
('py33', 'none', 'any')
('py32', 'none', 'any')
('py31', 'none', 'any')
('py30', 'none', 'any')

MartyG-RealSense, 2018-07-23:
May I just add that in Dorodnic's June 27 2018 RealSense webinar, he mentioned that it is easy to install Python via PyPy with $ pip install pyrealsense2 

[image]

closed 2018-07-23 by yiyunluo07

SarathVakacharla, 2019-01-16:
Is there any update to this yet?

> adding pip package for Mac OS

BenDavisson, 2019-08-29:
>adding pip package for Mac OS

Has there been an update for this issue in the last 8 months?  I have tried using pip install on python 3.7, 3.6, and 2.7  

All result in "ERROR: No matching distribution found for pyrealsense2"

MartyG-RealSense, 2019-09-11:
The most recent update I could find was March 2019, when it was reiterated that a pip install package for Mac still was not ready at that point in time.

https://github.com/IntelRealSense/librealsense/issues/3260#issuecomment-469361152

seanngpack, 2019-11-19:
any update on how to install on Mac?

vikram09, 2020-04-16:
Guys how do we install realsense2 on MAC for Python and OpenCV ,

seanngpack, 2020-04-16:
@vikram09 Check out #5275