realsenseai/librealsense #2120: C# NetStandard

jangernert, 2018-07-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC  |
| SDK Version                            | 2.14.0 | 
| Language                            | C# | 

### Issue Description
I noticed that with version 2.14.0 (probably introduced with 2.13, but I skipped that version) the build system of the C# wrapper was ported to CMake. Now there is a cmake option called `DOTNET_VERSION_LIBRARY` which is used to set `VS_DOTNET_TARGET_FRAMEWORK_VERSION`. But according to the cmake docs that only covers .NET Framework.
Before the supplied solution had a new style project file with multiple target frameworks configured, which also included `netstandard2.0`.
So my question comes down to: Is building the C# wrapper as a netstandard library currently possible? And if so, how would it be done?

Thank you =)

dorodnic, 2018-07-23:
Hi @jangernert 
Currently, I would use the project files from 2.12.0 with new code. Long term, CMake is better for the project. In the intermediate scope, we should find reasonable temporary solution. One suggestion was to keep the netstadard project files **in addition** to CMake. 
@matkatz1 for your consideration.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @jangernert,

do you need more support on this topic?

L0veGitHub, 2018-08-09:
Thank you for your kind help. I appreciate it.
I have just bought the D435 and I'm looking for information on Intel®
RealSense ™. I'm sorry if I inconvenienced you in some way, as I don't
understand the GitHub system yet.

jangernert, 2018-08-09:
No, I think the workaround is fine for now.
Should I close this ticket? Imo the ticket is still useful as a reminder that NetStandard builds are currently not working.

RealSense-Customer-Engineering, 2018-08-14:
[Realsense Customer Engineering Team Comment]
hi @jangernert,

yes, i think we can close this ticket while we are fixing issue, and other user can refer to this ticket even after ticket is closed.
and please feel free to open new ticket if any questions.

closed 2018-08-14 by RealSense-Customer-Engineering