realsenseai/librealsense #2121: HW sync undefined behaviour

ankarako, 2018-07-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D415 | 
| Firmware Version                   | 5.9.14| 
| Operating System & Version |   Win10| 
| Platform                                 | PC |
| SDK Version                            |  2.13 & 2.14                          | 
| Language                            |  C++                          | 
| Segment			| | 

### Issue Description
Hello, doing some experiments with HW syncing. 
* Does the HW sync imply Internal (Depth & Color Sensor) Sync? That is, are the depth-color streams of the same device synced? (I think that the answer is NO).
* Setting a device as MASTER (and forgetting about any other device) seems to have undefined behavior regarding color. I monitored the following:
    * The device does not produce any color frames
    * the device produces color and depth frames, but not at the correct rate (MASTER's clock)
    * The devices hangs.
What may cause that behavior?

Even if I set the device as master, and then open the rs-viewer, the color stream is not available. However, the depth stream seems to be OK.

Any help is really appreciated.
P.S.: Does HW sync concern only the depth sensor or the color too? If it does apply only for the depth sensor, is there a way to achieve Depth-Color Sync?

RealSense-Customer-Engineering, 2018-07-24:
[Realsense Customer Engineering Team Comment]
Hi

Here is how to enable the Sync feature on D415:
0) Refer to the attached picture;
1) RS-viewer must be version 2.13.0 or newer; Firmware version must be 5.9.14.0 or newer;
2) Depth and Color sensors must be set at same frame rate (i.e. 30fps )
3) "Inter Cam Sync Mode" set to "1.000"; (Default: 0.000; Master: 1.000; Slave: 2.000).
4) Color sensor, the "Enable Auto Exposure" must be Unchecked !!!
5) Keep in mind: Always start Depth camera first, then start the Color sensor; (Depth is master, Color is Slave)
6) To verify the Sync feature, see the number under icon "i"; they should be same. (see attached picture: "30.01" fps for both Depth and Color)

Thanks

ankarako, 2018-07-25:
Hello,

Thank you for your response. The steps I followed are just the same as the ones you mention (however there is no attachment). The device's behavior is still random. (I noticed that with the 2.14 rs-viewer the rgb sensor does not stream at all, but with the 2.13 sometimes it works as expected and sometimes hangs. Sometimes I get a Hardware Notification about a RT MIPI Error.)
The following is a link to a gif that captures my screen and presents the randomness I mentioned.
(The file was bigger than what github allows, so I uploaded it on imgur).

[rs-viewer 2.13 screen capture](https://imgur.com/a/p9q1ZfU)

RealSense-Customer-Engineering, 2018-08-03:
[Realsense Customer Engineering Team Comment]
Hi

After you turn off RGB camera (under Slave mode), please wait 5 seconds to let it fully discharge and then you can turn on RGB camera again. This usually can bring up RGB camera normally.

Thanks

zsyzgu, 2018-08-07:
Hello! Is the "Inter Cam Sync Mode" setting in the Realsense viewer? Or we should set it by code?

RealSense-Customer-Engineering, 2018-08-07:
[Realsense Customer Engineering Team Comment]
Hi

Are you asking Sync feature on a single camera or on multiple cameras?

Here is the Whitepaper regarding camera sync usage: see "Multi-camera_Configurations_whitepaper.PDF"
https://realsense.intel.com/intel-realsense-downloads/#whitepaper

Thanks

closed 2018-08-14 by RealSense-Customer-Engineering

pichu446, 2019-07-23:
Hi,

I have managed to have perfect time-synchronized flow (all RGB+Depth at 30 FPS) between two cameras with the realsense viewer app. But when I try to do the same without the app GUI, RGB wants always to be the master over depth (29.8 FPS instead of 30 FPS) and only the depth flows of the two cameras seems time-sync, not the rgb flows since the RGB and depth flow of the same device are not sync. I have tried to launch depth sensor before rgb sensor with sensor.start but without success. 

Is there any way to force the Depth sensor as the master over RGB?