realsenseai/librealsense #2122: failed to install librealsense2-utils

Rahul-Venugopal, 2018-07-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |                            | 
| Language                            |                         | 
| Segment			|                          | 

### Issue Description

I am following instructions as on the following link : 
[](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md)

I had done till the last step as per the instructions without any error , but when i run : `sudo apt-get install librealsense2-utils`
I get following 
### error 

**Err:1 http://de.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libegl1-mesa-dev amd64 17.2.8-0ubuntu0~16.04.1
  404  Not Found [IP: 141.30.62.24 80]
E: Failed to fetch http://de.archive.ubuntu.com/ubuntu/pool/main/m/mesa/libegl1-mesa-dev_17.2.8-0ubuntu0~16.04.1_amd64.deb  404  Not Found [IP: 141.30.62.24 80]**

Can anybody please help to solve this issue?

Thanks 
Rahul

RealSense-Customer-Engineering, 2018-07-25:
[Realsense Customer Engineering Team Comment]
Hello Rahul-Venugopal,

I am sorry you are experiencing this issue.  I have a few questions to help try to figure out what the issue may be.

-  Are you using a corporate internet connection, or any other connection that has a firewall in place?

-  Are you using latest instructions located here? https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

-  What Ubuntu OS are you using?  16.04, 16.04.5, 18.04?

>  We did not notice any issues downloading and installing packages from the above instructions.  If you are using an internet connection that is behind any type of firewall or has limited connection than issues such as not being able to retrieve packages can occur.  I suggest to try a different network connection and restart the above instructions from the first step and make sure no errors are seen.

Thank you.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
Hello again Rahul-Venugopal ,

Since there has been no response on this ticket for over a week, we will leave this issue open 7 more days.  If there is no response we will close this,

Thank you.

Rahul-Venugopal, 2018-08-10:
Hello ,

Sorry for the late response.

- I am using internet connection from University and also I tried with a private connection.

- Yes I am following the same instructions

- Ubuntu 16.04.4 LTS

When I tried from different network , I get the following error when running the command

 `sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u`

### Error 

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apt/cache.py", line 452, in update
    pulse_interval)
SystemError: E:Failed to fetch http://de.archive.ubuntu.com/ubuntu/dists/xenial/main/binary-arm64/Packages  404  Not Found [IP: 141.30.62.26 80], E:Failed to fetch http://de.archive.ubuntu.com/ubuntu/dists/xenial-updates/main/binary-arm64/Packages  404  Not Found [IP: 141.30.62.26 80], E:Failed to fetch http://security.ubuntu.com/ubuntu/dists/xenial-security/main/binary-arm64/Packages  404  Not Found [IP: 91.189.88.149 80], E:Failed to fetch http://de.archive.ubuntu.com/ubuntu/dists/xenial-backports/main/binary-arm64/Packages  404  Not Found [IP: 141.30.62.26 80], E:Some index files failed to download. They have been ignored, or old ones used instead.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 168, in <module>
    if not sp.add_source_from_shortcut(shortcut, options.enable_source):
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 782, in add_source_from_shortcut
    cache.update(sources_list=new_debsrc_entry.file)
  File "/usr/lib/python3/dist-packages/apt/cache.py", line 454, in update
    raise FetchFailedException(e)
apt.cache.FetchFailedException: E:Failed to fetch http://de.archive.ubuntu.com/ubuntu/dists/xenial/main/binary-arm64/Packages  404  Not Found [IP: 141.30.62.26 80], E:Failed to fetch http://de.archive.ubuntu.com/ubuntu/dists/xenial-updates/main/binary-arm64/Packages  404  Not Found [IP: 141.30.62.26 80], E:Failed to fetch http://security.ubuntu.com/ubuntu/dists/xenial-security/main/binary-arm64/Packages  404  Not Found [IP: 91.189.88.149 80], E:Failed to fetch http://de.archive.ubuntu.com/ubuntu/dists/xenial-backports/main/binary-arm64/Packages  404  Not Found [IP: 141.30.62.26 80], E:Some index files failed to download. They have been ignored, or old ones used instead.

RealSense-Customer-Engineering, 2018-08-11:
[Realsense Customer Engineering Team Comment]
Hello Rahul-Venugopal,

Thank you for the update.

I checked again on my end and I was able to connect to repository with the version of Ubuntu 16.04.4 with no issues.

This looks like it may still be a proxy issue.  You mentioned you are using an internet connection from within a University, most likely using their proxy(s).  Also you mention that you are using a private connection, did you disable all proxy settings when switching to your private internet connection?  .

Suggestions:
-  You can disable proxy settings by following this guide:  https://askubuntu.com/questions/251003/how-to-permanently-disable-http-proxy
-  If the above doesn't work, I suggest try to activate a wifi hotspot on your smart phone and connect to the wifi hotspot you created and try to install again - with proxies disabled.
-  If you are unable to disable proxies, I would reinstall Ubuntu 16.04 OS, here is a newer version of 16.04.5 to try:  http://releases.ubuntu.com/xenial/  Once a fresh copy of Ubuntu is installed, connect to the wifi hotspot you created or private network (not a network connected to University network backend at all) and try again.

Please let us know how this goes.
Thank you.

MohamedEhabHafez, 2019-03-31:
@RealSense-Customer-Engineering  I am facing problem with installing librealsense2-utils. My ubuntu is 32 bit, is it still not supported for 32bit and if yes, is there any other solution you recommend to use the realsense2-camera pkg??

gayanbrahmanage, 2020-02-10:
A possible reason could be adding a wrong  server to the list of repositories. Clear check of the list in 

`System Settings -> Software and Updates -> Other software`

Eg: `"deb http://realsense-hw............`

and Start over the installation ;)