realsenseai/librealsense #2123: avc: denied

green16, 2018-07-23:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  { D435 }  | 
| Firmware Version                   | { 05.09.02.00 } | 
| Operating System & Version |   {Win 10} | 
| Platform                                 | Asus Zenfone live  |
| SDK Version                            |  { legacy / 2.13.0 } | 
| NDK                                        |  {android-ndk-r16b } | 
| Segment			|  {Smartphone }                          | 

I just tried to build android Realsense_app following steps [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidJavaApp.md), but after launch nothing happens
my log:
07-23 16:30:37.274 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_dependencies_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_dependencies_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:37.563 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_0_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_0_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:37.630 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_1_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_1_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:37.693 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_2_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_2_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:37.768 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_3_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_3_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:37.831 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_4_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_4_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:37.893 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_5_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_5_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:37.954 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_6_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_6_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:38.014 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_7_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_7_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:38.076 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_8_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_8_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:38.135 1945-1945/com.example.librealsensetest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --debuggable --instruction-set=arm --instruction-set-features=smp,div,atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=krait --instruction-set-features=div --dex-file=/data/app/com.example.librealsensetest-2/split_lib_slice_9_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.librealsensetest-2@split_lib_slice_9_apk.apk@classes.dex) because non-0 exit status
07-23 16:30:38.158 1945-1945/com.example.librealsensetest I/InstantRun: starting instant run server: is main process
07-23 16:30:38.163 1945-1945/com.example.librealsensetest V/Monotype: SetAppTypeFace- try to flip, app = com.example.librealsensetest
07-23 16:30:38.165 1945-1945/com.example.librealsensetest V/Monotype:     Typeface getFontPathFlipFont - systemFont = default#default
07-23 16:30:38.717 1945-1945/com.example.librealsensetest W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
[... cut ...]
    Build Date: 04/17/16 Sun
    Local Branch: 
    Remote Branch: quic/LA.BF.1.1.3_rb1.13
    Local Patches: NONE
    Reconstruct Branch: NOTHING
07-23 16:30:39.231 1945-2211/com.example.librealsensetest I/OpenGLRenderer: Initialized EGL, version 1.4
07-23 16:30:39.590 1945-1945/com.example.librealsensetest W/art: Before Android 4.1, method int android.support.v7.widget.DropDownListView.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView

green16, 2018-07-23:
lsusb
Bus 001 Device 003: ID 8086:0ad6
Bus 001 Device 001: ID 1d6b:0002

ls -la /dev/bus/usb/001/
crwxrwxrwx root     usb      189,   0 2018-07-23 13:58 001
crwxrwxrwx root     usb      189,   2 2018-07-23 16:37 003

aarun2211, 2018-07-25:
Did you put the kernel in permissive mode? Try `setenforce 0` as root.

green16, 2018-07-25:
Yes I tried it

RealSense-Customer-Engineering, 2018-08-13:
[Realsense Customer Engineering Team Comment]
Please have a try the Android native samples follow this page.
https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md

The messages about "avc: denied" should be ignored, after run "setenforce 0" in root shell.

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
@green16  Any update for this issue? Did you try https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md  ?

RealSense-Customer-Engineering, 2018-08-28:
[Realsense Customer Engineering Team Comment]
@green16 Did you get chance to try https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md  Waiting for your update. Thanks!

RealSense-Customer-Engineering, 2018-09-04:
[Realsense Customer Engineering Team Comment]
@green16 Could you please update if you still have issue? Thanks!

RealSense-Customer-Engineering, 2018-09-11:
[Realsense Customer Engineering Team Comment]
@green16 Sorry that we didn't get response from you yet. Will close the ticket. If you still have question/issues please feel free to create another one. Thanks!

closed 2018-09-11 by RealSense-Customer-Engineering