realsenseai/librealsense #2124: Project depth image on color image python

yiyunluo07, 2018-07-23:
Hello, anyone who knows how to project depth image on color image on python?
I don't have camera but only two video files. One for color and one for depth(16 bit).
And now I need to align these two video files so that I can get the depth information with pixels in color video file.
What I wanna do is read the two video files frame by frame it is not possible to use wait_for_frames() and get_color_frame(), get_depth_frame()
Is there any method that can solve this problem? 
Thanks

yiyunluo07, 2018-07-23:
@dorodnic Sorry now I can only use frame not .bag file, how can I get the calibration of the device? And after that how can I align two streams?

dorodnic, 2018-07-23:
You would need to use `rs2_project_point_to_pixel`, `rs2_deproject_pixel_to_point` and `rs2_transform_point_to_point` methods to map every pixel from depth image to color image. 
To call these methods, you'd need `intrinsics` and `extrinsics`. Please see [API How-To](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-and-apply-depth-to-color-extrinsics), it would be pretty much the same in python.

yiyunluo07, 2018-07-23:
Hi @dorodnic ,
I'm still a little bit confusing about the way to get intrinsics and extrinsics. Must I have the camera connected and let it record or stream?

dorodnic, 2018-07-23:
You need the camera connected to get this data. You don't have to start streaming, but it is the easiest way to go about it. 
```python
import pyrealsense2 as rs;
p = rs.pipeline();
prof = p.start();
depth_intrin = prof.get_stream(rs.stream.depth).as_video_stream().get_intrinsics();
color_intrin = prof.get_stream(rs.stream.color).as_video_stream().get_intrinsics();
extr = prof.get_stream(rs.stream.depth).get_extrinsics_to(prof.get_stream(rs.stream.color));
```
We strongly encourage to use the `.bag` format, it was designed specifically to save users all this haste of serializing calibration tables, metadata, etc...

closed 2018-08-06 by yiyunluo07