realsenseai/librealsense #2125: librealsense bad image HResult=0x8007000B

atiato, 2018-07-23:
Hello ,

I'm compiling master librealsense under windows VC++ 2017 but I'm getting the following error when running the application cs-tutorial-1-depth

System.BadImageFormatException
  HResult=0x8007000B
  Message=An attempt was made to load a program with an incorrect format. (Exception from HRESULT: 0x8007000B)
  Source=Intel.RealSense
  StackTrace:
   at Intel.RealSense.NativeMethods.rs2_create_frame_queue(Int32 capacity, Object& error)
   at Intel.RealSense.FrameQueue..ctor(Int32 capacity) in C:\Users\omar_atia\Downloads\librealsense-master\librealsense-master\wrappers\csharp\Intel.RealSense\FrameQueue.cs:line 17
   at Intel.RealSense.Program.Main(String[] args) in C:\Users\omar_atia\Downloads\librealsense-master\librealsense-master\wrappers\csharp\cs-tutorial-1-depth\Program.cs:line 10

the dll is compiled within the source 

[image]

Configuration management is any CPU.
when I copy the dll from SDK release version Intel RealSense SDK 2.0 into debug directory it works, but library that we have after normal build for master branch doesn't work.

thanks,
Omar Atia

dorodnic, 2018-07-23:
The "Target Platform" for .NET examples should not be "Any CPU". 
For Win32, it should be x86 and for x64 - x64. This seems to be a bug with CMake for .NET

MartyG-RealSense, 2018-07-23:
Yes, I was about to say the same as Dorodnic, as I had a case with the old '2016 R3' RealSense SDK once that generated this error because of AnyCPU.

RealSense-Customer-Engineering, 2018-07-24:
[Realsense Customer Engineering Team Comment]
Which version of Camera, LibRealSense and FW are you using?

atiato, 2018-07-24:
I'm just compiling the code from master branch in Github, I didn't yet buy the camera.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @atiato,

do you need more support on this?