realsenseai/librealsense #2126: ZR300 not Recognized by Ubuntu 16.04

bsbretly, 2018-07-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |lll
| Camera Model                       | ZR300 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0-45-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |                           | 
| Language                            |  C++                       | 
| Segment			|  Robot                         | 

### Issue Description

Hello, I've installed ad built the v1.12.1 Tag of the librealsense library according to these instructions:
https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation.md

No matter which example I attempt to run, I get an error that there are no realsense devices are found. I've tried different USB ports/cables with no luck.

The output from dmesg:

[ 2289.820191] usb 1-1: new high-speed USB device number 6 using xhci_hcd
[ 2289.960696] usb 1-1: New USB device found, idVendor=0451, idProduct=8027
[ 2289.960702] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[ 2289.960706] usb 1-1: SerialNumber: 950E0069A8EF
[ 2289.961403] hub 1-1:1.0: USB hub found
[ 2289.961435] hub 1-1:1.0: 2 ports detected
[ 2290.080141] usb 2-1: new SuperSpeed USB device number 9 using xhci_hcd
[ 2290.100393] usb 2-1: New USB device found, idVendor=0451, idProduct=8025
[ 2290.100397] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 2290.101157] hub 2-1:1.0: USB hub found
[ 2290.101183] hub 2-1:1.0: 2 ports detected
[ 2290.388134] usb 2-1.2: new SuperSpeed USB device number 10 using xhci_hcd
[ 2290.409097] usb 2-1.2: LPM exit latency is zeroed, disabling LPM.
[ 2290.409842] usb 2-1.2: New USB device found, idVendor=8086, idProduct=0ad0
[ 2290.409846] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2290.409849] usb 2-1.2: Product: CX3-UVC
[ 2290.409852] usb 2-1.2: Manufacturer: Cypress
[ 2290.410696] uvcvideo: Found UVC 1.10 device CX3-UVC (8086:0ad0)
[ 2290.411582] uvcvideo 2-1.2:1.0: Entity type for entity Extension 3 was not initialized!
[ 2290.411585] uvcvideo 2-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[ 2290.411588] uvcvideo 2-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[ 2290.752447] usb 2-1.1: new SuperSpeed USB device number 11 using xhci_hcd
[ 2290.775934] usb 2-1.1: New USB device found, idVendor=8086, idProduct=0acb
[ 2290.775938] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2290.775940] usb 2-1.1: Product: Intel RealSense 3D Camera ZR300
[ 2290.775942] usb 2-1.1: Manufacturer: Intel Corp
[ 2290.775944] usb 2-1.1: SerialNumber: SN_4031801056
[ 2290.778784] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 2290.782214] uvcvideo 2-1.1:1.0: Entity type for entity Extension 2 was not initialized!
[ 2290.782217] uvcvideo 2-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 2290.783275] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 2290.786519] uvcvideo: Unable to create debugfs 2-11 directory.
[ 2290.786676] uvcvideo 2-1.1:1.2: Entity type for entity Camera 4 was not initialized!
[ 2290.787713] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 2290.791237] uvcvideo: Unable to create debugfs 2-11 directory.
[ 2290.791407] uvcvideo 2-1.1:1.4: Entity type for entity Processing 7 was not initialized!
[ 2290.791410] uvcvideo 2-1.1:1.4: Entity type for entity Camera 6 was not initialized!
[ 2290.791565] input: Intel RealSense 3D Camera ZR300 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.4/input/input19
[ 2312.943308] usb 1-1: USB disconnect, device number 6
[ 2313.089166] usb 2-1: USB disconnect, device number 9
[ 2313.089171] usb 2-1.1: USB disconnect, device number 11
[ 2313.142264] usb 2-1.2: USB disconnect, device number 10
[ 2315.849175] usb 1-1: new high-speed USB device number 7 using xhci_hcd
[ 2315.989508] usb 1-1: New USB device found, idVendor=0451, idProduct=8027
[ 2315.989531] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[ 2315.989534] usb 1-1: SerialNumber: 950E0069A8EF
[ 2315.990210] hub 1-1:1.0: USB hub found
[ 2315.990244] hub 1-1:1.0: 2 ports detected
[ 2316.109123] usb 2-1: new SuperSpeed USB device number 12 using xhci_hcd
[ 2316.129612] usb 2-1: New USB device found, idVendor=0451, idProduct=8025
[... 61 lines cut ...]
[ 2734.727768] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 2734.731414] uvcvideo 2-1.1:1.0: Entity type for entity Extension 2 was not initialized!
[ 2734.731418] uvcvideo 2-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 2734.732523] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 2734.735717] uvcvideo: Unable to create debugfs 2-17 directory.
[ 2734.735878] uvcvideo 2-1.1:1.2: Entity type for entity Camera 4 was not initialized!
[ 2734.736931] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 2734.740135] uvcvideo: Unable to create debugfs 2-17 directory.
[ 2734.740295] uvcvideo 2-1.1:1.4: Entity type for entity Processing 7 was not initialized!
[ 2734.740297] uvcvideo 2-1.1:1.4: Entity type for entity Camera 6 was not initialized!
[ 2734.740447] input: Intel RealSense 3D Camera ZR300 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.4/input/input21

Output from lsusb:

Bus 002 Device 002: ID 05ac:8406 Apple, Inc. 
Bus 002 Device 016: ID 8086:0ad0 Intel Corp. 
Bus 002 Device 017: ID 8086:0acb Intel Corp. 
Bus 002 Device 015: ID 0451:8025 Texas Instruments, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 05ac:8290 Apple, Inc. 
Bus 001 Device 003: ID 05ac:0274 Apple, Inc. 
Bus 001 Device 008: ID 0451:8027 Texas Instruments, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Thanks in advance for the help!

dorodnic, 2018-07-23:
Any warnings in the terminal when running the examples? 
What is the output if you change `rs::log_to_console(rs::log_severity::warn);` at [cpp-enumerate-devices.cpp:10](https://github.com/IntelRealSense/librealsense/blob/v1.12.1/examples/cpp-enumerate-devices.cpp#L10) to `rs::log_to_console(rs::log_severity::debug);` and rebuild?
All the system logs look Ok.

bsbretly, 2018-07-23:
Thanks for the response. When running the examples, I either get:

`RealSense error calling rs2_pipeline_start(pipe:0x1d1fad0):
    No device connected`
or 

`There are 0 connected RealSense devices.`

I made the change you suggest and rebuilt. When I try to run cpp-enumerate-devices in my /usr/local/bin foler with the command `./cpp-enumerate-devices` I get the following output in my terminal: 
./cpp-enumerate-devices: error while loading shared libraries: librealsense.so.1: cannot open shared object file: No such file or directory

dorodnic, 2018-07-24:
Something is not right - rs2_pipeline_start is not part of v1.12.1. Are you sure you are building the right version?

bsbretly, 2018-07-24:
Yes, but I installed the most recent version before realizing it doesn't support the ZR300. I then deleted the old repo and installed v.1.12.1. 

Should I delete the /usr/local/bin directory and try to build v.1.12.1 again?

dorodnic, 2018-07-24:
Regarding the kernel patch, I believe you can safely ignore it.

dorodnic, 2018-08-06:
@bsbretly, did you manage to make the camera work on your system?

bsbretly, 2018-08-06:
@dorodnic - thanks for following up. I did not - but have since moved on to a ROS wrapped version of the realsense library, which is suitable for my needs for now.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
@bsbretly
Do you still need help to deal with your problem? suggested you to use another clean machine to test with ZR300 and librealsense again, since from the thread looked like you didn't build the librealsense successfully.

bsbretly, 2018-08-09:
@RealSense-Customer-Engineering - I no longer require help, thanks!

closed 2018-08-09 by bsbretly