realsenseai/librealsense #2129: How to get rid of bad depths below min-range

xtrawurst, 2018-07-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |  Win10 | 
| SDK Version                            |  2.14.0 | 
| Segment			|  3D Scanning | 

We use the D415 for 3D Scanning and we frequently run into the issue of users getting too close (below min-range) with the camera, causing a lot of bad/stray points in the depth image, causing some corruption in the resulting model.

We were trying the D4 presets and the tuning parameters but unfortunately they do have a lot of side effects so, although we can theoretically find a setting that minimizes bad depths below min-range, it's not practical since we'd also be getting rid of most of the good depth values.

Hence, we're looking for a way to filter bad points from the depth image before they can corrupt the model. Is there some sort of confidence map that we could get through libRS to help with the filtering? Any other post-processing setting?

Btw, we appreciate all the existing post-processing options but in terms of importance, filtering bad depths below min-range and bad depths due to over-exposure (https://github.com/IntelRealSense/librealsense/issues/1912) would be **way** above the offered downsampling and spatial/temporal filtering.

MartyG-RealSense, 2018-07-24:
If your problem is with users getting too close to the camera lens, the advice in the link below may be of use.

https://communities.intel.com/message/553939#553939

xtrawurst, 2018-07-24:
Appreciate the quick response. However, reducing the input resolution is generally not a good option for us as it would come with undesired side-effects and would only move the problem to a closer range, instead of getting rid of it entirely.

MartyG-RealSense, 2018-07-24:
Some people have solved the problem of distance between camera and the scan target by mounting the camera overhead, pointing downwards.  Is this an option for the scans that you take?

xtrawurst, 2018-07-24:
Unfortunately no, in this case the camera is hand-held or mounted to a tablet that is.

MartyG-RealSense, 2018-07-24:
Section 5 of Intel's paper on camera tuning best-practices offers tips for minimum operating distance (MinZ) in section 5 of the paper.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

xtrawurst, 2018-07-24:
Thanks. Disparity shifts aren't going to work either unfortunately because, like resolution reduction, they only move the problem instead of solving it and they come with a hard trade-off.

MartyG-RealSense, 2018-07-24:
So could you confirm please that any solution that involves altering the resolution will not be suitable for you?

xtrawurst, 2018-07-24:
Anything that only slightly moves the problem instead of solving it doesn't work for us. Resolution reduction itself may be an option but right now it's simply not worth it if we look at the benefits vs. the side effects.

MartyG-RealSense, 2018-07-24:
I wonder if some kind of user feedback in the scanning application would help, such as an on-screen warning being displayed when MinZ was about to be exceeded,.  Or when the sensed distance between camera and object falls below a certain level, depth scanning stops.

xtrawurst, 2018-07-24:
Sure, yes. We already do that. Still can't avoid it completely though.

MartyG-RealSense, 2018-07-24:
Apologies that this process is taking a while.   It feels like we are making progress through a process of elimination though!

My sense is that D415 probably is an appropriate camera for this application, since new tablets from Panasonic and MobileDemand use the D410 and D415 cameras respectively for their scanning.  So it should work.  The problem is in minimizing bad data from proximity to the object.

AntoineWefit, 2018-07-24:
> Btw, we appreciate all the existing post-processing options but in terms of importance, **filtering bad depths below min-range** and bad depths due to over-exposure (#1912) would be way above the offered downsampling and spatial/temporal filtering.

Out of curiosity, have you set the "Depth Clamp Min" ? I find it super efficient to get rid of anything too close to my setup.

If I understood correctly, your application consists in scanning objects, which I assume might be close to the user ? If you could specify your scanning range, I might be able to share some of my settings for close range scanning (I'm using the max resolution btw). It is indeed hard to find a configuration that work well for both close and long range scanning.
I think it is worth trying to nicely set up the ".json" parameters before doing any extra post processing. Please let me know if you are interested.

MartyG-RealSense, 2018-07-24:
Thanks Antoine!  Yes, I was thinking of Depth Clamp Min / Max too, though wasn't sure if it would be one of the resolution-changing options that xtrawurst is keen to avoid.  

Thanks for your help!  Any insights you can offer would be welcome.

xtrawurst, 2018-07-24:
@AntoineWefit 
Thanks for your comment. "Depth Clamp Min" does not help because it only eliminates good depth points in the first place. The erroneous false positive depths caused by physically going below min-Z are typically quite a bit above min-Z.
So you'll find that if you get too close (below min-Z) to your setup you will get false positive depth values.

Our scanning range is 0.5 m to 3 m

MartyG-RealSense, 2018-07-24:
This will probably count as a stupid idea.   There is a thin line between stupidity and inspiration though, so I will take the chance.

If the tablet was mounted to some kind of short tripod that the user moved around in their hands, and the camera was attached to the tripod a bit behind the tablet in a way that didn't obscure its view, then it would be physically very hard for users to get the camera close enough to the object to go past MinZ, because the tablet would reach the object first.

[image]

AntoineWefit, 2018-07-24:
@xtrawurst 
I have to admit that I'm a bit confused
> Depth Clamp Min" does not help because it only eliminates good depth points in the first place. The erroneous false positive depths caused by physically going below min-Z are typically quite a bit above min-Z.
So you'll find that if you get too close (below min-Z) to your setup **you will get false positive depth values**.

Could you explain how/where did you find that ? I might have missed, or completely misunderstood something.

From what I tried, I understood that both "Depth Clamp Min/Max" are setting a range of depth for the scanning process. Anything below or upper those limits is basically ignored, and I didn't notice any loss of quality whatsoever on what is left in the scanning area (again, I'm not an expert, I only have experimented a few things).

Why not setting the "Depth Clamp Min" (yes, I like this parameter !) to 5000 (with "Depth Units" set to 100), so you are within your scanning range, and anything closer won't be processed by the camera ? You won't have any bad/erroneous points (in fact, you won't have any points at all).

I also think that the max achievable "depth perception" is in theory 0.1 mm, by setting the "Depth Units" to 100 (correct me if I'm wrong), thus, it may be hard to deal with artifacts below this size.

> Our scanning range is 0.5 m to 3 m.

I know that it is possible to scan objects close to 0.25 m with the D400 series (with decent quality), so 0.5 m is not that close and should not be a problem.

@MartyG-RealSense 
> it would be physically very hard for users to get the camera close enough to the object to go past MinZ, because the tablet would reach the object first.

Looks like a physical/hardware type of "Depth Clamp Min" to me :) in fact, a simple and efficient way to the problem !

xtrawurst, 2018-07-24:
@AntoineWefit What happens to the depth in your setup when you bring something (e.g. a large piece of cardboard) physically close to the camera (say 15 or 20 cm)?

AntoineWefit, 2018-07-24:
@xtrawurst I told you, the cardboard is completely ignored :) The data left aren't affected.
The field of view will obviously be blocked if the cardboard is very big, but the camera won't see it.
There will be a "shadow" of the cardboard projected on your object.

I could attach some pictures of a demonstrative setup tomorrow if you want to.

xtrawurst, 2018-07-24:
@AntoineWefit  Yes, if possible please post a video of RS Viewer running (showing the depth), where you bring a large piece of cardboard closer and closer to the camera. I'd be very interested in that.

AntoineWefit, 2018-07-25:
I'm sorry I can't provide a video, but you can easily reproduce these conditions yourself.

I simply put a cup in front of the object a want to scan (the box). The goal is to ignore anything too close to the camera (the cup), see by yourself : 

Here are some screenshots : 
"Situation 1" plastic cup in front of the box + no "Depth Clamp Min"
[image]
[image]
[image]

"Situation 2" plastic cup in front of the box + "Depth Clamp Min" set to 3400 (340 mm) to ignore the cup
[image]
[image]
[image]

As you see, there is no vertices in front of the box in "Situation 2", where the cup should be.

Be aware that you'll most certainly need to tinker the "Disparity Shift" to get the best quality if you modify  the "Depth Clamp Min/Max".

Let me know if that answered your question.

xtrawurst, 2018-07-25:
@AntoineWefit Thanks for providing the examples. Some questions:

* How close is the cup to the camera physically?
* What happens if you bring the cup closer (20 cm, 10 cm)?
* What is your disparity shift setting?
* What happens if you bring your RS camera really close to the white wall in the background (20 cm, 10 cm)?

jb455, 2018-07-25:
@xtrawurst 
You're not crazy, I see these effects too. As I bring my hand closer to the camera (streaming colourised depth on D415), the camera first 'blacks out' my hand at say 25cm, then if I move it closer still (to <20cm) it goes all speckly with patches of different colours on my hand. If I mouseover these patches it returns a variety of distances, 0.5m, 1m etc. This happens even if I put the clamp on. My hand will go black sooner because the camera sees it as too close, then when I move it closer it will light up again because it thinks the depth values are legit and greater than the clamp.
To partially get around this, we recognise when the subject first goes out of range, or gets too close to the near extreme, and tell the user to move backwards. Doesn't help if the user starts off too close though!

AntoineWefit, 2018-07-25:
@xtrawurst 
I dismantled the setup but : 
- Less than 340 mm, so I'd say it was at around 250 mm.
- The "shadow" of the cup projected on; the scanned object and everything behind the cup; will be bigger and will occult a bigger part of the scenery. (Have to check if what @jb455 said happens too)
- You can check the values on the screenshots.
- Depends if you change the parameters or not : 
1. Same parameters : I think that, with the "Depth Clamp Min" set up to 3400 (so 34 cm), you won't see anything on the depth stream if you come closer than that.
2. Different parameters : let's say no "Depth Clamp Min"; your camera will still see the scenery (depending on many other parameters) until it reaches the minimal working distance. And at this point, you won't get any good data obviously.

You should really try this yourself, because it's not that difficult to set up and I'm not sure that my explanations are clear enough.

@jb455 
I might had experienced that too (can't remember precisely though), but I think that also happened when being excessively close to the camera. I will try to check that out.

xtrawurst, 2018-07-25:
@AntoineWefit Thanks. Sorry, hadn't noticed the parameter settings on the left.

I'm sure what's helping you in your case is the big disparity shift of 100, not the "depth clamp min". 
I'm also sure you'll see a lot of bad depths if you reset the disparity shift to 0 or if you try to see past, say, 1m in depth.

As I said before a disparity shift wouldn't solve the problem for us as our scanning range is up to 3m.

MartyG-RealSense, 2018-07-25:
Dropping back into this excellent conversation for a moment ...   :)

Regarding the 'tablet and camera on a tripod' concept above: the 400 Series cameras' tripod attachment screw-thread on the bottom of the camera is apparently a standard thread-size for tripods.  So it might not be too difficult to screw the RealSense camera onto the top of the tripod, behind the tablet, without needing a special mount for the camera.

https://communities.intel.com/message/524440#524440

Intel's own documentation shows them using a tripod other than the one supplied with the camera.  They state: "Any medium sized tripod should be sufficient.  In this example setup, we used a Manfrotto Compact Light Aluminum Tripod with ball head.  The ball head makes
adjustment to device orientation easier".

[image]

[image]

AntoineWefit, 2018-07-25:
@xtrawurst 
> I'm sure what's helping you in your case is the big disparity shift of 100, not the "depth clamp min".

Yes and no, I found that both impact the data quality quite a lot.

> I'm also sure you'll see a lot of bad depths if you reset the disparity shift to 0 or if you try to see past, say, 1m in depth.

Yes, I think so ! As I said before, it is difficult to find parameters that will suit a big range of depth.

> As I said before a disparity shift wouldn't solve the problem for us as our scanning range is up to 3m.

I didn't test anything at that distance, so I can't say anything on that, sorry.

@MartyG-RealSense 
Something like that maybe ? (maybe without the overkill-clamp-thingy, but you get the idea).

[image]

xtrawurst, 2018-07-26:
@RealSense-Customer-Engineering 
Any ideas w.r.t. the original question?

closed 2018-11-08 by dorodnic

srinath-iko, 2019-07-30:
Hey @RealSense-Customer-Engineering @RealSenseCustomerSupport  I am facing the same problem as @xtrawurst 
I have the depth clamp set too. Lowering the resolution or changing the disparity shift would just shift the error. I have attached the pictures describing the error. As an object(my hand here) comes closer to the camera, there seems to be many wrong depth values calculated and shifted behind as you can see in the second image.This seems to be an inherent problem with the type of imaging I guess as the SR300 does not cause such errors, being a coded-light sensor camera.

Is there any solution to this issue provided by realsense or would I have to do a bit of post-processing?

[image]

[image]

MartyG-RealSense, 2019-07-30:
As your hand becomes sufficiently close to the camera to pass under its minimum depth, the image will start progressively breaking up as the hand gets closer and closer to the lens.  Using a disparity shift would normally allow the hand to get closer.  I understand though why this is not an option for you (other users have opted not to use a disparity shift because they do not want to change the data).

Section 8b on page 6 of Intel's tuning guide PDf offers advice about using the camera at very close range and explains why matching between the left and right imagers can break down at this range with stereo cameras.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

AntoineWefit, 2019-07-30:
> Is there any solution to this issue provided by realsense or would I have to do a bit of post-processing?

@srinath-iko Did you try to tune the _**Min Distance**_ of the _**Threshold Filter**_ ? You'll have to enable the post-processing in the Viewer : go to _Stereo Module / Post-Processing / Threshold Filter / Min Distance_. It should get rid of every data below that value of distance.
[image]

I think that it could be a solution for you :)
Hope this'll help !

srinath-iko, 2019-07-31:
@MartyG-RealSense 
Thank you for the suggestion. The depth unit size wouldn't change the inherent errors caused. It again just shifts the error.

@AntoineWefit 
Thank you for the input. I tried the post-processing threshold too. Although this gets rid of the data below the minimum distance, if an object goes near, the depth points are still present at the wrong location as they are calculated to be behind the threshold minimum.

xtrawurst, 2019-07-31:
@srinath-iko @RealSense-Customer-Engineering 
I concur. The issue is that the D400 series cannot disambiguate false positive depth values from true positive depth values well. There are some tuning parameters for the depth algorithm but typically these reduce false positive depths and true positive depths equally, so not always a good trade off.

What would be required is a clean-up algorithm in the driver or on the chip that's designed solely to get rid of false positive depth values. The old venerable Primesense PS1080 had this and I believe Occipital is doing this in the driver for their new Structure Core sensor. Both with very good results.
It's a big, big missing feature in the, otherwise excellent, RealSense offering that I still hope is going to get filled at some point.

MartyG-RealSense, 2019-07-31:
@xtrawurst The lack of a 'confidence threshold' in the 400 Series to decide if a value is valid or invalid is a consequence of the stereo depth technique.  The SR300, which is based on coded light / structured light, does have confidence threshold support.  @dorodnic comments on the subject here:

https://github.com/IntelRealSense/librealsense/issues/3185#issuecomment-460540961

srinath-iko, 2019-08-01:
@MartyG-RealSense @xtrawurst 
Yes, I did check @dorodnic 's comments. its an inherent issue with the method of imaging that can only be solved with a few custom post processing steps.

Thanks!