realsenseai/librealsense #2133: Source Code for distinguish between Photo and human being in realsense streaming

atiato, 2018-07-24:
Hi all,

I need to know how to code or which API to depend on to differentiate between a photo and human being when detecting & recognizing faces using 3rd party along with latest C# Wrappers or librealsense native code .

I don't want to use old intel realsense SDK Version 2 which is no longer supported,   I want to use this github sources to do such functionality .

Do you have any advice ?

thanks,
Omar Atia

MartyG-RealSense, 2018-07-24:
The RealSense SDK Version 2, if you mean SDK 2.0, is supported.  It is in ongoing development and is now up to version 2.14.0.

The RealSense D415 camera has been shown to work with face recognition software called CloudCompare.

https://communities.intel.com/message/549475#549475

atiato, 2018-07-24:
The RealSense SDK Version 2, if you mean SDK 2.0, is supported. It is in ongoing development and is now up to version 2.14.0.

if version 2.14.0 is supported but samples doesn't have face detections or face recognition functions ? 

I was looking on CloudCompare link you provided and sources in their github but I found rendering into 3D objects Only , so where are the C sharp or C++ code that is related to face detection and recognition ? note in detection and recognition we need to differetiate between photo and human being , could you just give me some hints so I can dig into and analyze .

thanks,
Omar Atia

dorodnic, 2018-07-24:
Please note that the purpose of this project is to provide the best integration of our hardware, and not any sophisticated algorithms. In today's CV ecosystem, there are so many great solutions for these types of problems, some open source, some not, some cross platform, some not. 
The most natural way to approach this problem is to detect faces in 2D via OpenCV ([using Haar Cascades](https://docs.opencv.org/3.3.1/d7/d8b/tutorial_py_face_detection.html)).
Once you have the bounding box for the face, you'd need a classifier that could distinguish between a face and a fake. The simplest one can be plane-fitting and looking at depth variance. More reasonable approach is to train a deep-learning classifier with some fake and some real data. 
While this classification problem is hard in 2D, with depth your neural network will instantly pick up on the differences.

closed 2018-08-07 by RealSense-Customer-Engineering

bingws, 2019-03-11:
@dorodnic Hello! You mentioned the method of biopsy detection using depth information: "The simplest one can be plane-fitting and looking at depth variance. More rational approach is to train a deep-learning classifier with some fake and some real data." Are there any simple examples? I want to learn. Thank you.

MartyG-RealSense, 2019-03-11:
Perhaps the software **TensorFlow** would help you to achieve your goal.

https://www.tensorflow.org/

It has a tutorial on training a system to tell the difference between real and fake images.

https://www.tensorflow.org/alpha/tutorials/generative/dcgan