realsenseai/librealsense #2134: Ubuntu 18.04 build issue with libgtk-3-dev & libssl-dev on D435 libRS 2.14.0

evercoast-smm, 2018-07-24:
Hi,

I'm seeing the following problems when building librealsense-2.14.0 on a fresh Ubuntu 18 machine:

~/Src/RealSense$ uname -r
**4.15.0-29-generic**

Errors:

**~/Src/RealSense$ sudo apt install libgtk-3-dev
Reading package lists... Done
Building dependency tree      
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
libgtk-3-dev : Depends: libgdk-pixbuf2.0-dev (>= 2.30.0) but it is not going to be installed
                Depends: libpango1.0-dev (>= 1.40.5) but it is not going to be installed
                Depends: libcairo2-dev (>= 1.14.0) but it is not going to be installed
                Depends: libfontconfig1-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.**

And:

**~/Src/RealSense$ sudo apt install libssl-dev
Reading package lists... Done
Building dependency tree      
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
libssl-dev : Depends: libssl1.1 (= 1.1.0g-2ubuntu4) but 1.1.0g-2ubuntu4.1 is to be installed
              Recommends: libssl-doc but it is not going to be installed**

Any advice greatly appreciated!! Thanks!

ev-mp, 2018-07-26:
Hello @evercoast-smm ,
It seem that the distribution has some missing/broken packages.
You can run `dpkg -l | grep ^..r ` to list those, and then use Ubuntu tools to rectify it 
`sudo apt update --fix-missing` and  
`sudo apt install -f <package_name>`

evercoast-smm, 2018-08-05:
Thank you very much! This resolved the issue!

jorgept, 2019-03-22:
hi,

i have the same problems.
Can someone help me?

in Nginx machine with ubuntu 18.04 installed

root@vps445139:~# uname -r
**4.15.0-46-generic**

`
root@vps445139:~# sudo apt install libgtk-3-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtk-3-dev : Depends: libglib2.0-dev (>= 2.49.4) but it is not going to be installed
                Depends: libgdk-pixbuf2.0-dev (>= 2.30.0) but it is not going to be installed
                Depends: libpango1.0-dev (>= 1.40.5) but it is not going to be installed
                Depends: libatk1.0-dev (>= 2.15.1) but it is not going to be installed
                Depends: libatk-bridge2.0-dev but it is not going to be installed
                Depends: libcairo2-dev (>= 1.14.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
`
And:

`root@vps445139:~# sudo apt install libssl-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libssl-dev : Depends: libssl1.1 (= 1.1.0g-2ubuntu4.3) but 1.1.1a-2+ubuntu16.04.1+deb.sury.org+1 is to be installed
              Recommends: libssl-doc but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
`

Thank you so mutch

juandarango, 2020-09-04:
Hello I just found a way to solve the problem! 
fisrt I tryed with:  
```
sudo apt update
sudo apt update --fix-missing
```
that solved some dependencies, but not all, so I installed **aptitude** (sudo apt-get install aptitude) and then used it to install the realsense library. 
`sudo aptitude install <your library>`
it suggested one solution, but it didn't work, so I pressed no (n) and suggested another solution, that required to downgrade some packages. And the installation finished successfully. Downgrade packages under your own risk!

evercoast-smm, 2020-09-04:
Thank you, we're good on this issue. Can close.

Evercoast <http://www.evercoast.com>  |  Sebastian Marino  |  Co-founder &
CTO