realsenseai/librealsense #2135: Camera stuck in wait for frames

aarun2211, 2018-07-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0                                         | 
| Platform                                 | Android  |
| SDK Version                            |  2                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                         | 

### Issue Description
Hello, 
I am streaming the realsense color and depth channels (640x480, fps = 6) to a Google Pixel XL, via a usb 3.0 OTG cable. realsense_app displays the color and depth frames sometimes, but most of the time it is stuck in the wait_for_frames function. I would like to understand why this is happening. I have tried other resolutions and frame rates as well, but it gives the same result. I face this issue if I only enable the depth stream as well. Any insights will be appreciated. 
Thank you.

dorodnic, 2018-07-25:
Hi @aarun2211 
Could you please share librealsense logs? You can enable them via [environment variable](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#q-how-do-i-enable-librealsense-logs) or by adding `rs2::log_to_console(RS2_LOG_SEVERITY_DEBUG);` to example source code.
Also, can you double check that the device is in fact recognized as USB3? You can see it in `rs-enumerate-devices`.

aarun2211, 2018-07-30:
Hi @dorodnic 
Thank you for the help and apologies for the late reply. I tried to enable the librealsense logs but it didn't work. Could you give me more information on how to do this on Android device? 
Yes, the device recognizes it as a USB3.0 (Dev ID: 8086:0ad3). I have attached the logcat, and have also included some log messages with tags RealSenseJava (messages from MainActivity.java) and RealSenseNDK (messages from native-lib.cpp). As you can see from the logcat, line 166, the code is stuck at wait_for_frames, and does not proceed. 
[logcat.txt](https://github.com/IntelRealSense/librealsense/files/2242507/logcat.txt)

Thank you.

fotsing365, 2018-08-16:
Hi everyone. 
I 'm facing same problem. after editing frame size from 480X270 to any format the application hangs on a black frame. I tried to find out what happen using logcat messages and I noticed that app execution don't cross the instruction `profile = pipe.start(config);` in the file native-lib.cpp

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
@aarun2211 & @fotsing365
Please try to run the command line sample (`rs-depth` and `rs-enumerate-devices`) on Android.
https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md

@fotsing365 , would you share more detail information about hardware and software? including kernel's version.

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
Would you share the commit ID of your kernel?

AFAIK, in some system, the CPU is too busy to receive a complete frame in UVC.
and in other hand, librealsense will drop incomplete frame.
https://github.com/IntelRealSense/librealsense/blob/v2.15.0/src/linux/backend-v4l2.cpp#L835

You can try to add __android_log_print in backend-v4l2.cpp Line 840. Make sure the libreaslsense is still working, when no new frame coming.

fotsing365, 2018-08-20:
Hi @RealSense-Customer-Engineering  
Camera D435
Rooted Lenovo YT3-850F
Android 6
SDK 2.12.0
Ubuntu 16.04

alex1122a, 2018-09-21:
Dear @RealSense-Customer-Engineering,
I am facing the same issue on Android. Depth and color streams are coming fine, however once per 3~4 times, the program gets stuck in pipe.wait_for_frames(). Once stuck, it requires several app restarts to get things working again.
Could you provide any steps to debug or fix this issue?

RealSense-Customer-Engineering, 2018-10-10:
[Realsense Customer Engineering Team Comment]
pls reference to the following Android example to troubleshooting the issue:

https://github.com/zhouwg/irsa/tree/master/examples/rs_multicam

btw, pls use the latest librealsense source code(add log mechanism in Android for troubleshooting sw issues)

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes

RealSense-Customer-Engineering, 2018-10-15:
[Realsense Customer Engineering Team Comment]
@aarun2211,

pls reference to:

https://github.com/zhouwg/irsa/blob/master/docs/IRSA_introduction.txt

https://github.com/zhouwg/irsa/blob/master/how-to-build-irsa.txt

https://github.com/zhouwg/irsa/blob/master/examples/how-to-build-examples.txt

https://github.com/zhouwg/irsa/blob/master/examples/rs_multicam/app/src/main/cpp/native-lib.cpp#L457

 can we close the ticket accordingly? Please update. Thanks!

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
@aarun2211 Any other questions about this ticket? Looking forward to your update. Thanks!

aarun2211, 2018-10-30:
@RealSense-Customer-Engineering 
Thank you for the help. I will look into these resources and resolve the issue. I will close the issue accordingly.

closed 2018-10-30 by aarun2211