realsenseai/librealsense #2137: How to transfrom pixel coordinates in a color stream to the pixel coordinates in the depth stream avoiding rs2::align?

AkhmedAnam, 2018-07-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I have a rectangle in a color image and I wanna get this rectangle for the depth image. Now I use **_rs2::align_** object with **_process_** method. But it is very costly (probably because this method align all image while I need only 4 pixels). Is it possible to find coordinates of a rectangle got for a color image in the corresponding depth image? (maybe using intrinsic/extrinsic). I had an idea to transform color pixel corrdinates to 3D color coordinates then to 3D depth coordinate and finnaly this 3D depth coordinates to 2D depth coordinates. But it seems impossible to get 3D color points from 2D color points because the **_rs2_deproject_pixel_to_point(float point[3], const struct rs2_intrinsics * intrin, const float pixel[2], float depth)_** function needs the depth parameter.

RealSense-Customer-Engineering, 2018-07-30:
[Realsense Customer Engineering Team Comment]
@AkhmedAnam
There is no closed formula/transformation defined to align 2D images directly.
There are ways to tackle this issue. The canonical method , imho, is to apply CV algorithms based on SIFT, like OpenCV, still other ways.

Since inside the LibRS, we focus on delivering depth data, it requires depth frame to be one side of the alignment(from <-> to).

RealSense-Customer-Engineering, 2018-08-07:
[Realsense Customer Engineering Team Comment]
@AkhmedAnam
Answered your question?

yuemasy, 2018-08-16:
@RealSense-Customer-Engineering 

I know a pixel (i,j) in the unaligned color frame and want to know the pixel's 3d coordinates (x,y,z).
 "unaligned color frame" I mean the original color frame that hasn't been aligned.

I think the process should be a pixel (i,j) in the unaligned color frame -->a pixel in aligned color frame->a pixel in depth frame, then I can know the pixel's 3d coordinates (x,y,z). 

As you said , "There is no closed formula/transformation defined to align 2D images directly.", you mean a pixel in the unaligned color frame can't directly transform to a pixel in the aligned color frame. 

Then you said "it requires depth frame to be one side of the alignment(from <-> to).",  Can it be unaligned color frame  ->depth frame? is it possible?

Thank you very much.

dorodnic, 2018-08-21:
Hi @yuemasy 
We have added a [pull-request](https://github.com/IntelRealSense/librealsense/pull/2265/files#diff-cc67aaf99bc160428474034159ab6c8cR117) to enable color pixel to depth pixel mapping, it will be available in next release.

> One more question, if depth frame is aligned to color frame, then can I calculate 3d point based on the aligned depth frame? According to my experiment, the answer is No. I want to confirm with you.

At the moment no, but we are going to make a fix later today to allow it. Also planned for upcoming release. 

Thank you for your feedback, it is important to us.

yuemasy, 2018-08-22:
hi  @dorodnic ,
Thanks for reply. Looking forward to the new release.

dorodnic, 2018-11-29:
I forgot to update - [rsutil.h:116](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L116) lets you map color pixel to depth.

dorodnic, 2019-02-04:
I believe can be closed

closed 2019-02-04 by dorodnic